nodemcu/nodemcu-firmware #2165: init.lua:x: out of memory

philuser, 2017-11-04:
### Expected behavior
server view
### Actual behavior
__init.lua:2: out of memory__

> dofile("init.lua")
init.lua:2: out of memory
stack traceback:
	[C]: in function 'listen'
	init.lua:2: in main chunk
	[C]: in function 'dofile'
	stdin:1: in main chunk
> 

### Test code
a test with a basic program on a web page

```Lua
srv=net.createServer(net.TCP,30) 
srv:listen(80,function(conn) 
  conn:on("receive",function(conn,payload) 
    print(payload) 
    conn:send("Hello, NodeMCU.")
    end) 
end)
```
### NodeMCU version

NodeMCU custom build by frightanic.com
	branch: master
	commit: 443e8219527f5b2190324a969a4586f9d3d731bf
	SSL: false
	modules: file,gpio,net,node,pwm,tmr,uart,wifi
 build 	built on: 2017-11-04 11:58
 powered by Lua 5.1.4 on SDK 2.1.0(116b762)

### Hardware
NodeMCU "Amica"

marcelstoer, 2017-11-08:
One for #1010 I guess...

closed 2017-11-08 by marcelstoer

rptnscz, 2018-09-28:
Guess I have seen the same problem after upgrade to 2.2.1-master_20180915.
It is believed, that the solution was erase flash with 
esptool.py --port <serial-port-of-ESP8266> erase_flash
and flash the firmware again.