nodemcu/nodemcu-firmware #2166: why esp32-dev version not have tmr.delay() && tmr.now() method

oupula, 2017-11-05:
RT.
how to get cycle time?
tmr.create().register(...) ?

> When and why should I avoid using tmr.delay()?
If you are used coding in a procedural paradigm then it is understandable that you consider using  tmr.delay() to time sequence your application. However as discussed in the previous section, with NodeMCU Lua you are coding in an event-driven paradigm.

If you look at the app/modules/tmr.c code for this function, then you will see that it executes a low level  ets_delay_us(delay). This function isn't part of the NodeMCU code or the SDK; it's actually part of the xtensa-lx106 boot ROM, and is a simple timing loop which polls against the internal CPU clock. It does this with interrupts disabled, because if they are enabled then there is no guarantee that the delay will be as requested.

tmr.delay() is really intended to be used where you need to have more precise timing control on an external hardware I/O (e.g. lifting a GPIO pin high for 20 μSec). It will achieve no functional purpose in pretty much every other usecase, as any other system code-based activity will be blocked from execution; at worst it will break your application and create hard-to-diagnose timeout errors.

The latest SDK includes a caution that if any (callback) task runs for more than 10 mSec, then the Wifi and TCP stacks might fail, so if you want a delay of more than 8 mSec or so, then using  tmr.delay() is the wrong approach. You should be using a timer alarm or another library callback, to allow the other processing to take place. As the NodeMCU documentation correctly advises (translating Chinese English into English): tmr.delay() will make the CPU work in non-interrupt mode, so other instructions and interrupts will be blocked. Take care in using this function.

oupula, 2017-11-05:
/**
  * @brief  Get system time, unit: microsecond.
  *
  * This function is deprecated. Use 'gettimeofday' function for 64-bit precision.
  * This definition will be removed in a future release.
  */
uint32_t system_get_time(void)  __attribute__ ((deprecated));

。。。

marcelstoer, 2017-11-07:
@TerryE announce quite a while ago that he wanted to deprecate `tmr.delay()`, see #1892. So, it probably makes sense to not include it in the ESP-32 branch.

devsaurus, 2017-11-12:
Also see [Johny's initial module description](https://github.com/nodemcu/nodemcu-firmware/issues/1619#issue-191839813) for reasons `tmr.now()` wasn't ported:
> Wall-clock time should not be dealt with in this module. This module is to be only for relative-time alarms.

closed 2017-11-12 by marcelstoer