nodemcu/nodemcu-firmware #2168: Feature request: mDns client

Schwenkner, 2017-11-06:
Is it possible to add mDns client support to the firmware?

My use case is to locate ip of the MQTT broker in my network. The broker is running in the local net and the nodemcu clients have to determine the ip of the broker. The device where the broker is running I can implement the mdns server.

marcelstoer, 2017-11-20:
> Is it possible to add mDns client support to the firmware?

I'm sure it is (not being a firmware programmer) but basically this is one for #1010. @pjsg wrote the [NodeMCU mDNS _server_ module](https://nodemcu.readthedocs.io/en/latest/en/modules/mdns/). So, if at all he'd be a likely candidate to add mDNS _client_ support to NodeMCU.

closed 2017-11-20 by marcelstoer

SimoGira, 2018-07-01:
maybe [this](https://github.com/udaygin/nodemcu-mdns-client) can help you