nodemcu/nodemcu-firmware #2170: wifi.sta is a nil value

aj22r, 2017-11-11:
### Expected behavior
The ESP8266 will connect to a wifi access point
### Actual behavior
"stdin:1: attempt to index field 'sta' (a nil value)"
### Test code
```Lua
> wifi.setmode(wifi.STATION)
> wifi.sta.config("ssid", "pass")
stdin:1: attempt to index field 'sta' (a nil value)
> print(wifi.sta.status())
stdin:1: attempt to index field 'sta' (a nil value)
>
```
### NodeMCU version
1.0.20150120

### Hardware
Wemos D1 mini

marcelstoer, 2017-11-11:
We don't support ancient versions, see #1010

closed 2017-11-11 by marcelstoer

aj22r, 2017-11-12:
But this is the newest version according to [http://nodelua.org/doc/download_firmware/](http://nodelua.org/doc/download_firmware/)

djphoenix, 2017-11-12:
@aj22r [NodeLua](https://github.com/haroldmars/nodelua) is another project, and seems like it's not updated in last 3 years.

marcelstoer, 2017-11-12:
@haroldmars hasn't been active on GitHub in a long time and the last entry on his blog is from April 2014, I guess it's save to assume that NodeLua is dead.