nodemcu/nodemcu-firmware #2178: Flashing latest firmware to 512kB modules

RafNie, 2017-11-19:
### Expected behavior
Update documentation so that not hold back developers before using latest firmware on 512kB. Because they can benefit from it.

### Actual behavior
There is statement in the [NodeMcu documentation](https://nodemcu.readthedocs.io/en/master/en/lua-developer-faq/):
_Note that the 1.5.4.1-final branch is the last available release if you still wish to develop applications for 512Kb modules_
Similar note is also on the cloud build service.
I suppose that this information is obsolete or at least confusing for lua developers. I've performed comparison between firmware versions and it occurs that the latest firmwares remains more free space for user code. The bin file size is smaller for master despite compiling with the one additional module (ds18b20).

I dont know how it would looks for SDK 2.0.0, but I supose it may be merit of SDK 2.1.0. Also, maybe there were some other optimizations in the master code. Hard to guess. It's just my speculation
 

### Test code
Test procedure: flashing firmware and checking remaining flash memory for Lua code 

Test 1.
	1.5.4.1 final
	Total : 15813 bytes
	Used  : 0 bytes
	Remain: 15813 bytes

Test 2.
	master commit: 443e8219
	Total : 46937 bytes
	Used  : 0 bytes
	Remain: 46937 bytes

Uploaded lua application seems work well on both tested firmwares.

### NodeMCU version
**1.5.4.1 final**, float, 10 modules: dht,file,gpio,http,net,node,ow,tmr,uart,wifi
Bin file size: 466804

**Latest master** build, float, commit: 443e8219527f5b2190324a969a4586f9d3d731bf, 11 modules: dht,ds18b20,file,gpio,http,net,node,ow,tmr,uart,wifi
Bin file size: 428624

### Hardware
Tested on ESP-O1 module

devsaurus, 2017-11-19:
Thanks a lot @RafNie for sharing your findings!
They sound plausible when considering [Johny's statement](https://github.com/nodemcu/nodemcu-firmware/issues/1810#issuecomment-299838933) while we updated to 2.1.0. I can imagine that the function sections thing also decreased ROM footprint since unused code doesn't go into the image anymore.

jmattsson, 2018-01-03:
That's a welcome surprise - does anyone have time to PR this?

PR #2222 (2018-01-10): Remove guidance to fall back to 1.5.4.1 for 512Kb modules
Fixes #2178.

- [x] This PR is for the `dev` branch rather than for `master`.
- [x] This PR is compliant with the [other contributing guidelines](https://github.com/nodemcu/nodemcu-firmware/blob/dev/CONTRIBUTING.md) as well (if not, please describe why).

As reported by @RafNie, a firmware image from the current dev branch fits quite nicely in a 512Kb flash. I've ran some comparisons which second his request to remove the respective note in the Lua developer FAQ:

## Default module selection

| branch  | `_flash_used_end` | `0x10000.bin [... cut]
closed 2018-01-10 by marcelstoer