nodemcu/nodemcu-firmware #2179: Feature request: sequence number field in probe request frame

mstfbirinci, 2017-11-20:
### Missing Feature
NodeMCU firmware support us two fields in event monitor callback function(wifi.eventmon.AP_PROBEREQRECVED) as MAC and RSSI. I need to handle another frame field `Sequence Number`.

### Justification
Because of identify the per probe request in each bursty.

### Workarounds
There is no any workaround solution.

pjsg, 2017-11-20:
I am working on a promiscuous mode listener for nodemcu. It will give you the entire PROBE frame. However, the AP/STATION functionality is disabled at that time. Would this solve your problem?

mstfbirinci, 2017-11-20:
AP/STATION functionality is disabled means it cannot connect an access point ?

pjsg, 2017-11-21:
It depends on exactly what you want to do... if you can receive a number of probes and then bring up the AP, then you can send that data -- but you can't do it at the same time (as far as I know).

mstfbirinci, 2017-11-21:
In the current sitation, it is listening probe request on AP_STATION mode and send over wifi with mqtt protocol at the same time. I only want to handle other fields in the probe request.

mstfbirinci, 2017-11-21:
I have checked Espressif NON-OS SDK 2.1.0 api reference and realized probe_request event only support MAC and RSSI. I think I need to report this to the Espressif.

mstfbirinci, 2017-11-23:
According to IEEE 802 standarts, we can not handle management frames in promiscuous mode. Probably, promiscous mode you're working on will not give probe frame.

pjsg, 2017-11-23:
You definitely do get the management frames (at least, you get the first 112 bytes of them). This is enough to be able to get probe request frames and extract useful data from them.

mstfbirinci, 2017-11-23:
Mentioned about this mode at given article below as 

> Even in promiscuous mode, an 802.11 adapter will only supply packets to the host of the SSID the adapter has joined. Although it can receive, at the radio level, packets on other SSID's, it will not forward them to the host.

[Difference - Promiscuous vs. Monitor Mode ](http://lazysolutions.blogspot.nl/2008/10/difference-promiscuous-vs-monitor-mode.html)

As far as I understood about promiscuous mode, it makes SSID filtering at radio link level. So we can not give all probe request except for packets that came with same SSID joined network or broadcasts.

closed 2019-08-04 by stale[bot]