nodemcu/nodemcu-firmware #2180: ESP32: Anybody want to collaborate on dual core support?

chilipeppr, 2017-11-21:
I would love to get dual core support going in Lua for ESP32. Anybody want to collaborate? I know how to code, I'm just not sure where to get started.

TerryE, 2017-11-25:
NodeMCU Lua is a non-blocking I//O (at least as the network is concerned) event driven framework, very similar in many ways to **node.js**.  Having an extra core running Lua isn't going to increase the throughout of your application. The Lua implementation is non-reentrant and whilst in principle standard Lua can be compiled to support multiple OS threads, there is a code size and performance hit doing this.

So this is a _lot_ of work for no gain. Bad idea. Sorry.

chilipeppr, 2017-11-25:
Well, there are a couple of situations I've run into where I've craved some
isolation of one core from the other. When I'm driving stepper motors, I
notice pausing when other activity is happening due to the event driven
nature. If I'm pulsing lasers and toggling to different PWMs for power I
notice some delays here and there as well which I attribute to events
getting processed. If I'm trying to count pulses on my hall effect sensor
on my CNC spindle, I lose count when events are holding things up. So, the
thought is, if you could run a block of code on the 2nd core kind of like
how Javascript lets you dump off an async task now, then that would be
awesome.

TerryE, 2017-11-25:
Two cores won't help you here. There is a separate PR to add pulse counting to the GPIO interrupt handler, which would help.

chilipeppr, 2017-11-25:
The pulse counting would help one scenario, so that would be cool. Are you
saying that NodeMCU Lua does actually use both cores automatically on a
task by task basis? Or does it just use one core and the other one is idle
all the time? I figured the 2nd core was just sitting idle.

TerryE, 2017-11-25:
No, the Lua application is single threaded and only runs on one core at any time. But there are lots of other tasks executing under the realtime OS, such as those supporting the network stack and I/O. They can run on the second core without blocking Lua as is the case on the ESP8266.

As I said this would be a huge chunk of complex  work. You aren't going to get the developers who have the skills to do this for no material benefit.  There are far more important low hanging fruit.

devsaurus, 2019-05-31:
Not much activity lately. Feel free to re-open in case of concrete activity.

closed 2019-05-31 by devsaurus