nodemcu/nodemcu-firmware #2181: What does "E: M xxxxx" mean?

wayne3377, 2017-11-21:
<------------------------ BUG REPORT -----------------------------------------
### Expected behavior
esp 8266 boot up successful.

### Actual behavior
Waiting for IP address...
Wifi connection is ready! IP address is: 10.30.10.118
Connect to Server will resume momentarily, you have 3 seconds to abort.
Waiting...
E:M 1576
E:M 1576
E:M 1576
E:M 1576
E:M 1576
E:M 1576
E:M 1576
E:M 1576
MQTT Connect failed reason: -1
MQTT Connect failed reason: -5

### Test code
Lua

### NodeMCU version

NodeMCU2.1.0

### Hardware
 8266-12F

wayne3377, 2017-11-21:
Sometimes, it will raise error: "not enough memory" when I call subscript function. It is difficult to reproduce, so I didn't catch the screen shot. Any more detail information about memory limitation?

marcelstoer, 2017-11-26:
Another one for #1010

closed 2017-11-26 by marcelstoer