nodemcu/nodemcu-firmware #2183: timers can only be started once

pjsg, 2017-11-21:
### Expected behavior

The code should print out 'on' and then 'off' whenever the pulse() method is invoked. From my reading of the documentation, the timer should remain registered and it should be startable multiple times. 

### Actual behavior

The first time it works fine, and the second time it never prints 'off'

### Test code

```Lua
pulsetimer = tmr.create()
pulsetimer:register(100, 0, function() 
    print('off')
  end)

function pulse()
  print ('on')
  pulsetimer:stop()
  pulsetimer:start()
end```

### NodeMCU version
Current dev branch
### Hardware
wemos d1 mini

devsaurus, 2017-11-21:
Isn't mode 0 actually `tmr.ALARM_SINGLE`? That would explain the observed behavior. 
You probably want mode `tmr.ALARM_SEMI`.

pjsg, 2017-12-02:
Indeed. It is probably a bug that the :start() method doesn't throw an exception if it doesn't actually start a timer!

stefan-muc, 2017-12-26:
Documentation of tmr.start() says: https://nodemcu.readthedocs.io/en/latest/en/modules/tmr/#returns_9

> Returns true if the timer was started, false on error

So I guess it's up to you to check the return code.

marcelstoer, 2018-01-23:
@pjsg do you intend to change the `:start()` behavior or can this be closed?

pjsg, 2018-01-23:
Thanks for reminding me. I was thinking that the failure to start a timer should be an exception, but I wonder if that will break anybody's code....

pjsg, 2018-01-28:
Actually, after looking at the tmr code, I think that it just makes sense to leave the code the way that it is.

closed 2018-01-28 by pjsg