nodemcu/nodemcu-firmware #2185: spi.setup() does not work on Wemos d1 mini

ThePiGrepper, 2017-11-25:
8<------------------------ BUG REPORT -----------------------------------------
### Expected behavior
> spi.setup(1, spi.MASTER, spi.CPOL_LOW, spi.CPHA_LOW, 8, 8)
//spi configured and no error

### Actual behavior
> spi.setup(1, spi.MASTER, spi.CPOL_LOW, spi.CPHA_LOW, 8, 8)
//t hangs in there and does nothing, after a couple seconds it reboots.

### Test code
> spi.setup(1, spi.MASTER, spi.CPOL_LOW, spi.CPHA_LOW, 8, 8) --just this, with any clk_divider

*I enabled debug mode and use some prints to see where it stopped and it does during spi_master_init(), around the spi_set_clkdiv(spi_no, clock_div) function in app/driver/spi.c.

### NodeMCU version
Tested on:
commit 22e1adc4b06c931797539b986c85e229e5942a5f
and more recently:
commit: 443e8219527f5b2190324a969a4586f9d3d731bf

### Hardware
WEMOS D1 mini.
8<------------------------ END BUG REPORT -------------------------------------

ThePiGrepper, 2017-11-25:
I've tried this with the two wemos D1 mini I got ( they might actually be clones, but they work fine otherwise) , both present the same issue and they are basically unmodified. Maybe it's just a misconception on my part, but the same exact code worked on a esp12E I have, and this are ESP12S  believe.

devsaurus, 2017-11-25:
That's really kind of strange. The symptoms would suggest that the gpio pins used by spi are tied or connected externally somehow. But I know that this shouldn't be the case on a bare Wemos D1 mini (I use these a lot myself).

Can you please double check that there's nothing connected to SCLK, MOSI, MISO,  and SS pins?

ThePiGrepper, 2017-11-25:
I have the schematics for the wemos d1 mini board and it doesn't appear so. but ofc that doesnt mean anything so I'll check the board directly and confirm this. As I said, these boards might be clones so maybe they have some differences.

ThePiGrepper, 2017-11-27:
I fixed the issue. Apparently, the issue was caused by flashing the firmware by using esptool with the -ff option with "20m" instead of the default 'keep' option. At first this was needed, because flashing it with the default option made the lua interpreter throw garbage no stop. Now, after flashing it again this time w/o the -ff option, everything worked out the usual way. I dont quite understand how this could fix the issue, maybe I'm not understanding correctly how the -ff option works. I'd like to know how this happened but I think I should close this issue now.

closed 2017-11-27 by ThePiGrepper