nodemcu/nodemcu-firmware #2187: QMC5883L support next to HMC5883

infrat, 2017-11-29:
### Missing feature
QMC5883L support next to HMC5883

### Justification
Because many cheap magnetometer modules on EBAY or ALIEXPRESS utilise QMC5883 instead of HMC5883 it would be nice to support both in nodemcu. Both modules are similar, but they have different registers structure and some additional adjustments must be done to have it working.

I've created my own version of /app/modules/hmc5883l.c working correctly with QMC sensors, but currently don't have an idea how to deal with supporting both modules in NodeMCU - create separate modules in lua for QMC and HMC, or make existing HMC library "smart" and recognize what type of chip to use. That's the open question.
I'm going to fork nodemcu-firmware, and share my changes. Also appreciate if you could help me with finding proper way of integrating it with current firmware.

### External resources
HMC5883L datasheet: https://cdn-shop.adafruit.com/datasheets/HMC5883L_3-Axis_Digital_Compass_IC.pdf
QMC5883L datasheet: http://wiki.epalsite.com/images/7/72/QMC5883L-Datasheet-1.0.pdf

Main differences between both chips:
- I2C address (QMC: 0x0D vs HMC: 0x1E)
- different ODRs (QMC: 10Hz, 50Hz, 100Hz, 200Hz vs HMC: 0.75Hz, 1.5Hz, 3Hz, 7.5Hz, 15Hz, 30Hz, 75Hz)
- data register address (QMC: 0x00:05 vs HMC: 0x03:08)
- data order (QMC: 0x00.XLSB, 0x01.XMSB, 0x02.YLSB, 0x03.YMSB, 0x04.ZLSB, 0x05.ZMSB vs HMC: 0x03.XMSB, 0x04.XLSB, 0x05.ZMSB, 0x06.ZLSB, 0x07.YMSB, 0x08.YLSB)
... and few more

pjsg, 2017-11-29:
My personal feeling is to make one module that can handle both. This will maximize the chance that the user will have the right software module built into their image for the hardware module that came from aliexpress!

infrat, 2017-11-29:
Cool. I'll try to publish fork with my PoC in few hours.

closed 2019-08-04 by stale[bot]