nodemcu/nodemcu-firmware #2194: File chunck

bastos1505, 2017-12-11:
When I tried to read a line bigger than 256 bytes file.readline() doesn't works. The function return just 256 first bytes 

8<------------------------ BUG REPORT -----------------------------------------
### Expected behavior
If a line have up to 1024 bytes, the function file.readline() should return all the content up to find "\n"

### Actual behavior
The function return only the first 256 bytes instead 1024(MAX)
### Test code
Supposing to have any file with a line content of 512 bytes length.
var LINE = file.read()  -- This should return 512 byte + "\n", but return only the 256 first bytes

The Bug is in the function file_readline( lua_State *L) in file.c
Instead to use: return file_g_read(L, **LUAL_BUFFERSIZE**, '\n', fd);
Should use: return file_g_read(L, **FILE_READ_CHUNK**, '\n', fd);

```Lua
-- add code here
```
### NodeMCU version
Which branch are you on? If you know the Git revision then add it here as well.
Branch master Rev 2.1.0

### Hardware
Describe which ESP8266 device you use and document any special hardware setup 
required to reproduce the problem.

8<------------------------ END BUG REPORT -------------------------------------

8<------------------------ FEATURE REQUEST ------------------------------------
### Missing feature

### Justification
Tell us why you would like to see this feature added.

### Workarounds
Are there any workarounds you currently have in place because the feature is missing?

8<------------------------ END FEATURE REQUEST --------------------------------

pjsg, 2017-12-17:
That would certainly increase the size up to 1024, but it isn't clear to me why that it is a better limit. Why not read until the character is found (or memory is exhausted)?

bastos1505, 2017-12-17:
In my case, I need read more than 256 bytes per line. 
Doing that modification, the function readline match with documentation. Otherwise, the documentation is wrong.

pjsg, 2017-12-17:
I think that you are right -- you do want to have some level of protection against input data. Just go ahead and submit a pull request for that change.

marcelstoer, 2017-12-18:
Just for the sake of completeness here's what [the documentation](https://nodemcu.readthedocs.io/en/latest/en/modules/file/#filereadline-fileobjreadline) says:

> Lines are defined as zero or more bytes ending with a EOL ('\n') byte. If the next line is longer than 1024, this function only returns the first 1024 bytes.

Thanks @bastos1505, looking forward to a PR.

marcelstoer, 2018-02-24:
@bastos1505 can we expect a PR from you any time soon or shall we close this?

closed 2018-03-06 by marcelstoer