nodemcu/nodemcu-firmware #2195: NodeMCU Wikipedia Page

TerryE, 2017-12-13:
The tag "NodeMCU" is confusing in that is refers to two separate things:
- The H/W Devkit originally marketed by Zeroday.  AFAIK, you can still pick up clearance items, but I think that production has been killed off because of the cheaper Wemos boards. 
- This Github project.

The WP page largely talks about this software project, but still refers to Vowstars holding page at **nodemcu.com** as the official website in the callout info box and shows a Devkit picture.  However all of the other relevant links in the content (other than the "official website" link) referred to this project's pages.

Unfortunately over this last year, the WP editors have been going through harmonising pages styles and having a general blitz on external links other than in references.  As part of this purge, all links to this project (other than one change by @devsaurus) have been removed.  As one editor [said](https://en.wikipedia.org/wiki/User_talk:Stesmo#NodeMCU) "if the official site doesn't link to documentation/github, then why would we?)"

So I do think that since the original hardware is now obsolete and no longer manufactured and the nodemcu.com page unchanged in about 3 years, we should move the focus of the wikipage to this project.

Thoughts?  See also my post on the article's talk page.

marcelstoer, 2017-12-17:
I've been unhappy about that WP for almost as long as I've been involved here. Yet, I never found the energy and time to do much about it. In the last two years I only redacted a few obviously wrong statements and made sure the code samples are more or less in line with the current API.

The question of whether the term "NodeMCU" refers to the devkits or this project is as old as the respective projects. When I started with https://nodemcu.readthedocs.io/ I first asked @nodemcu whether this shouldn't rather be nodemcu-_firmware_.readthedocs.io instead. As you can see he then preferred "NodeMCU" to refer to the firmware rather than the devkits.

Terry, what would it take for https://en.wikipedia.org/wiki/NodeMCU to redirect to https://en.wikipedia.org/wiki/NodeMCU_(disambiguation) which would then list firmware and devkit pages?

As for the website I was in contact with @nodemcu in June because I asked him to fix a code example. He then told me about https://nodemcu.github.io/ which is served from https://github.com/nodemcu/nodemcu.github.io. He offered to redirect the .com domain to that site if someone were to put together a pull request. We'd just need two static pages in English and Chinese. I had then planned to do that myself but life took a different turn.

TerryE, 2017-12-17:
Marcel, I will send you an offline email, because some data aren't in the public domain, so I can't use them here.

The editor that's triggering this is being a PITA, but for what he feels are the right reasons, and some of them I share.  What is driving this is an attempt to standardise the style of all WP articles, and to stop articles being created primarily as link bate.  To this end, the [External Links](https://en.wikipedia.org/wiki/Wikipedia:External_links) policy has been set up.  It was first created in 2006, but over the years, they've tightened up and enforce this a lot more -- to the extent that now a page should normally only have one in article external link to the the "official website".  This being said, the S/W infobox also has a link to the repository, so I suggest that we swap out the HW aspects to history and focus the article on the firmware.

The potential issue with doing this is that the editor who is doing this "link police" will revert what we are doing if he view is as controversial because he will view the Github committers as having a conflict of interest and therefore biased.  So step on is really to get the holding page correct, and them dig up the necessary web references to underpin the content.  The WP rules are that we should really use content from other reputable IT sources to underpin the article content to avoid an AfD application on the grounds of lack of notability.

limbo666, 2017-12-29:
Hi guys I would like to bring tom your attention another issue regarding the "nodemcu" term.
Searching the internet is a real pain. Lot of users are searching for nodemcu examples and search engines are pointing to arduino examples because of the hardware board. 
This is very frustrating for nodemcu (lua) users. I don't know if is there any possible trick to change the project name or identifier but i think it would be great to solve this issue.

chilipeppr, 2017-12-29:
I agree with you. For ESP32 what I now search for instead of nodemcu is
something like "esp32 lua websocket". Even for the ChiliPeppr workspace
names this name issue became a big deal. For ESP8266 the workspace used to
be called NodeMCU, but then for ESP32 I needed a unique name so I went with
"ESP32 for Lua" but then went and renamed the NodeMCU workspace to "ESP8266
for Lua" with NodeMCU in parentheses.

[image: Inline image 1]

vowstar, 2017-12-30:
@TerryE  I'm not the owner of nodemcu.com, zeroday is the real owner.
I'm one of the developer of nodemcu firmware, and NodeMCU devkit 0.8, 0.9, 1.0.

limbo666, 2018-09-03:
@TerryE & @marcelstoer 
I know that this post is about the website and WP but as you may know "nodemcu" term is lost to the hardware. 
What I mean is the search results on google and bing are pointing almost always to arduino projects, examples and other hardware related nodemcu projects. For the average use it is a real pain to search for examples and help on the internet.
   
I don't know if it possible to change the project name (I think it as a great solution to this problem though), but from my point of view the situation needs some action. I think that the job is done here on nodemcu firmware is undervalued under this situation.

limbo666, 2018-09-05:
@jmd13391 I understand your point, but this is not 100% correct. Try to search for _"nodemcu lua push notifications"_ for instance and check the results you getting from Google. 
Search engines are guided by the most uncommon word in the phrase to produce results. In out case the lua term is almost ignored.

In any case I'm not here to argue. My opinion is that this is a drawback for new users. If a user gets a ton of results on Arduino and is struggling to find lua examples then soon or later he is going to abandon nodemcu firmware.

TerryE, 2018-09-05:
> In our case the lua term is almost ignored.

Putting quotes around _lua_ makes it a mandatory search term.  But what we really need are more external high-ranked sites / articles referring back here.

cpapir, 2019-06-10:
I can not even find information how to connect the shield to the Aduino Uno. Any one out there can help?

TerryE, 2019-06-10:
@cpapir, see #1010. This is the wrong list to ask this type of Q.

TerryE, 2020-04-19:
@marcelstoer @vowstar this one is still dangling.  I am not sure how we can progress this, but if we can't then we shouldn't leave it open.

vowstar, 2020-04-19:
@TerryE I've no problem to purge/clean up outdated information of WP. 
I have not been involved in the NodeMCU project for a long time. This project is full of various memories, thank you for your hard work.

Occasionally found two photos (Maker Faire Shenzhen 2015)
[image]
[image]

At the time, we were still producing NodeMCU hardware, and all the profits were used to donate NodeMCU development boards, print NodeMCU promotional pages, and participate in exhibitions.
A lot of things happened later, so that the trademark of NodeMCU is also held by Ai-thinker, and Ai-Thinker has nothing to do with us. (Although he has been producing various new hardware whose name includes NodeMCU). Later, I made up my mind to enter the chip industry and began to design RISC-V related chips. This is another story. Maybe one day, NodeMCU can run on the chip I designed.

Thank you very much again!

marcelstoer, 2020-04-19:
Great to hear back from you Ray!

>  the trademark of NodeMCU is also held by Ai-thinker

Now that's interesting, I wasn't aware of that (or I forgot).

> Although he has been producing various new hardware whose name includes NodeMCU

So far I simply assumed anyone - incl. Ai-thinker - who sells boards that have "NodeMCU" in their name would be guilty of trademark infringement.

closed 2021-06-20 by stale[bot]