nodemcu/nodemcu-firmware #2197: MQTT re-connect issue

Questdoino, 2017-12-17:
**Expected behavior**
ESP8266 should reconnect to MQTT server even if server went through power reset

**Actual behavior**
If the MQTT server got down because of hybernation(laptop) or power reset (Raspberry Pi) ESP's behavior after reconnecting is weird. It receives the CONNACK frame and returns ACK, but after 10 seconds keep alive period it should send PINGREQ instead of this it closes the socket. The ESP's ip address is 192.168.2.155 in this example
[image]
If server was down and up using console everything seems to be ok.
[image]

**Test code**
```
server_ip = "192.168.2.9"
server_port = 1883
client_id = wifi.sta.getmac()
m = mqtt.Client(client_id, 10, "MY_LOGIN", "MY_PWD")

mqtt_offline_flag = 1
mqtt_connection_attempts = 0

mqtt_timer = tmr.create()
mqtt_timer:register(10000, tmr.ALARM_SEMI, function() do_mqtt_connect() end)
mqtt_timer:start ()

function handle_mqtt_error(client, reason) 
  uart.write (0, "MQTT error - ") 
  print (reason)
  if (reason < 0) then mqtt_timer:start() end
end

function do_mqtt_connect()
  uart.write (0, "Connection attempt to mqtt # ")
  print (mqtt_connection_attempts)
  mqtt_connection_attempts =  mqtt_connection_attempts + 1
  m:connect(server_ip, server_port, 0, function(client)
    print ("mqtt_conn")
    mqtt_offline_flag = 0
  end,
  handle_mqtt_error)
end

m:on("offline", function(client) 
    mqtt_offline_flag = 1
    print ("Mqtt offline")
    mqtt_timer:start()
end)
```

**NodeMCU version**
2.1.0 dev and master seems like working the same

**Hardware**
Mosquitto v1.4.14 for Debian Stretch on x64 laptop
Mosquitto v1.4.14 for Raspbian (Debian Jessie) on Raspberry Pi 3
ESP8266 (ESP-01) the smallest chinese module from Aliexpress

andreykeen, 2018-01-27:
I have the same problem.

After connecting I close access to the broker through iptables (that would not turn off raspberry pi 3).
ESP loses connection. Then I remove iptables drop rules and ESP connected to the server. But every _keepalive_ seconds ESP again loses connection.

If I shutdown mosquitto server with `systemctl stop mosquitto` ESP writes to console: 

> MQTT event: broker going to offline

Then I do `systemctl start mosquitto` and ESP connecting to brocker and then everything OK

**Test code**
```Lua
local timer_mqtt_connect = tmr.create()
local keepalive = 30
local m = mqtt.Client(conf.general.hostname, keepalive, "", "", 1)

m:on("message", function(client, topic, data)
    print ("MQTT event: message")
end)

local function mqtt_connected(client)
    print("Connected to MQTT broker "..conf.mqtt.server..". "..tmr.time())
end

local function mqtt_error(client, reason)
    print("Connect to "..conf.mqtt.server.." failed. Reason: "..reason..". "..tmr.time())
    client:close()
    timer_mqtt_connect:start()
end

function do_mqtt_connect()
    print("Trying to connect to "..conf.mqtt.server..". "..tmr.time())
    m:connect(conf.mqtt.server, 1883, 0, 0, mqtt_connected, mqtt_error)
end

timer_mqtt_connect:register(5*1000, tmr.ALARM_SEMI, function()
    do_mqtt_connect()
end)

timer_mqtt_connect:start()
```

**NodeMCU version**
2.1.0-master_20170824
Lua 5.1.4 on SDK 2.1.0(116b762)

**Hardware**
Mosquitto v1.4.10 for Raspbian (Debian Jessie) on Raspberry Pi 3
ESP8266 Wemos D1 mini

andreykeen, 2018-02-01:
I did some experiments.
I took last [commit](https://github.com/nodemcu/nodemcu-firmware/tree/f87d68ff8f14128e18be4061123b644120ed9387) in dev branch and build firmware (I used [esp-open-sdk](https://github.com/pfalcon/esp-open-sdk))

Then I again tested the behavior of the mqtt module with this lua code:
```lua
local timer_mqtt_connect = tmr.create()

local m = mqtt.Client("nodemcu-test", 20, "", "", 1)

m:on("offline", function(client)
    print ("MQTT: Broker going to offline")
    client:close()  -- I tried with and without it
    timer_mqtt_connect:start()
end)

local function mqtt_connected(client)
    print("MQTT: Connected")
end

local function mqtt_error(client, reason)
    print("MQTT: Connecting failed. Reason:"..reason)
    client:close()  -- I tried with and without it
    timer_mqtt_connect:start()
end

local function do_mqtt_connect()
    print("MQTT: Trying to connect. Heap:"..node.heap().." Uptime:"..tmr.time())
    m:connect("192.168.2.250", 1883, mqtt_connected, mqtt_error)
end

timer_mqtt_connect:register(5*1000, tmr.ALARM_SEMI, function()
    do_mqtt_connect()
end)

timer_mqtt_connect:start()
```

I found same problem on [Stackoverflow](https://stackoverflow.com/questions/48335112/mqtt-clientconnect-not-established-callback-is-unexpectedly-called-after-a) and enabled NODE_DBG for functions:
mqtt_socket_disconnected
mqtt_socket_reconnected
mqtt_socket_connected
mqtt_socket_connect

### First experiment when I close access to the server with iptables (emulation of network problems or sudden server shutdown).
Terminal output (my comments begins --)

>  -- **Start and normal connect**
> MQTT: Trying to connect. Heap:35232 Uptime:56
> enter mqtt_socket_connect.
> leave mqtt_socket_connect.
> enter mqtt_socket_connected.
> leave mqtt_socket_connected.
> MQTT: Connected
> 
> 
> --  **On server (192.168.2.250) I close port 1883 with iptables DROP**
> enter mqtt_socket_disconnected.
> MQTT: Broker going to offline
> leave mqtt_socket_disconnected.
> 
> MQTT: Trying to connect. Heap:33184 Uptime:240
> enter mqtt_socket_connect.
> leave mqtt_socket_connect.
> MQTT: Connecting failed. Reason:-5
> 
> MQTT: Trying to connect. Heap:35000 Uptime:250
> enter mqtt_socket_connect.
> leave mqtt_socket_connect.
> MQTT: Connecting failed. Reason:-5
> 
> MQTT: Trying to connect. Heap:35000 Uptime:260
> enter mqtt_socket_connect.
> leave mqtt_socket_connect.
> MQTT: Connecting failed. Reason:-5
> 
> 
> -- **After three attempts I opened port 1883 on server**
> MQTT: Trying to connect. Heap:35000 Uptime:270
> enter mqtt_socket_connect.
> leave mqtt_socket_connect.
> enter mqtt_socket_connected.
> leave mqtt_socket_connected.
> MQTT: Connected
> 
> 
> --  **Then unexpected behavior begins**
> enter mqtt_socket_reconnected.  -- **Why this function is called?**
> MQTT: Connecting failed. Reason:-5
> enter mqtt_socket_disconnected.
> MQTT: Broker going to offline
> leave mqtt_socket_disconnected.
> leave mqtt_socket_reconnected.
> 
> MQTT: Trying to connect. Heap:35176 Uptime:296
> enter mqtt_socket_connect.
> leave mqtt_socket_connect.
> enter mqtt_socket_connected.
> leave mqtt_socket_connected.
> MQTT: Connected
> 
> enter mqtt_socket_reconnected.  -- **Why this function is called?**
> MQTT: Connecting failed. Reason:-5
> enter mqtt_socket_disconnected.
> MQTT: Broker going to offline
> leave mqtt_socket_disconnected.
> leave mqtt_socket_reconnected.
> 
> MQTT: Trying to connect. Heap:35176 Uptime:322
> enter mqtt_socket_connect.
> leave mqtt_socket_connect.
> enter mqtt_socket_connected.
> leave mqtt_socket_connected.
> MQTT: Connected
> 
> enter mqtt_socket_reconnected.  -- **And again**
> MQTT: Connecting failed. Reason:-5
> enter mqtt_socket_disconnected.
> .......
> .......

It repeats indefinitely.

### Second experiment when I shutdown mosquitto on the server with "systemctl stop mosquitto.service"

> -- **Start and normal connect**
> MQTT: Trying to connect. Heap:35776 Uptime:12
> enter mqtt_socket_connect.
> leave mqtt_socket_connect.
> enter mqtt_socket_connected.
> leave mqtt_socket_connected.
> MQTT: Connected
> 
> 
> -- **systemctl stop mosquitto.service**
> enter mqtt_socket_disconnected.
> MQTT: Broker going to offline
> leave mqtt_socket_disconnected.
> 
> MQTT: Trying to connect. Heap:35224 Uptime:623
> enter mqtt_socket_connect.
> leave mqtt_socket_connect.
> enter mqtt_socket_reconnected.
> MQTT: Connecting failed. Reason:-5
> enter mqtt_socket_disconnected.
> leave mqtt_socket_disconnected.
> leave mqtt_socket_reconnected.
> 
> MQTT: Trying to connect. Heap:35136 Uptime:628
> enter mqtt_socket_connect.
> leave mqtt_socket_connect.
> enter mqtt_socket_reconnected.
> MQTT: Connecting failed. Reason:-5
> enter mqtt_socket_disconnected.
> leave mqtt_socket_disconnected.
> leave mqtt_socket_reconnected.
> 
> MQTT: Trying to connect. Heap:35136 Uptime:634
> enter mqtt_socket_connect.
> leave mqtt_socket_connect.
> enter mqtt_socket_reconnected.
> MQTT: Connecting failed. Reason:-5
> enter mqtt_socket_disconnected.
> leave mqtt_socket_disconnected.
> leave mqtt_socket_reconnected.
> 
> 
> -- **After three attempts I run server (systemctl start mosquitto.service)**
> MQTT: Trying to connect. Heap:35136 Uptime:639
> enter mqtt_socket_connect.
> leave mqtt_socket_connect.
> enter mqtt_socket_connected.
> leave mqtt_socket_connected.
> MQTT: Connected

This is normal behavior.

**This behavior does not allow to implement a stable IoT device for a smart home.**

chathurangawijetunge, 2018-02-04:
Does anyone have any solutions for this issue.? How to fix it

andreykeen, 2018-02-05:
Until the bug is fixed, you can use something like this:

```Lua

local is_connected = false
local timer_mqtt_connect = tmr.create()
local m = mqtt.Client(conf.general.hostname, 20, "", "", 1)

local function restart()
    -- [[ Temporary solution of a bug #2197 ]]
    -- [[ Restart ESP if disconnect occurred after connecting ]] --
    print ("RESTART CHIP  RESTART CHIP  RESTART CHIP  Time:"..tmr.time())
    node.restart()
end

m:on("offline", function(client)
    print ("MQTT: Broker going to offline")
    is_connected = false
    restart()
end)

local function mqtt_connected(client)
    print("MQTT: Connected")
end

local function mqtt_error(client, reason)
    print("MQTT: Connecting failed. Reason:"..reason)

    if is_connected then
        restart()
        return
    end
    
    is_connected = false
    timer_mqtt_connect:start()
end

local function do_mqtt_connect()
    print("\nMQTT: Trying to connect. Heap:"..node.heap().." Uptime:"..tmr.time())
    m:connect(conf.mqtt.server, 1883, mqtt_connected, mqtt_error)
end

timer_mqtt_connect:register(5000, tmr.ALARM_SEMI, function()
    do_mqtt_connect()
end)

timer_mqtt_connect:start()

```

chathurangawijetunge, 2018-02-05:
Thanks bro....

chathurangawijetunge, 2018-02-05:
Unfortunately in my situation I can't reset the device.
I guess I have to wait till the bug is fixed :-(

FrankX0, 2018-02-08:
@chathurangawijetunge: this will fix this re-connection issue.

closed 2018-02-09 by marcelstoer

chathurangawijetunge, 2018-02-09:
Will the could build will also have the bug fixed..?

marcelstoer, 2018-02-09:
The fix was merged into the `dev` branch. If you build a binary from that branch it will include the fix.

And please don't cross-post your comments everywhere.

chathurangawijetunge, 2018-02-09:
Reconnecting issue is fixed.... Thanks 4 that.

But for every reconnection we loos heap memory.
Just disconnect the internet connection and check after some time for every connection and disconnect we loos memory till device crashes

andreykeen, 2018-02-09:
Show your lua code.

chathurangawijetunge, 2018-02-09:
sorry My bad..... 
i was checking the new Freeware  (master branch)
dev is working fine.
many thanks again

PR #2256 (2018-02-09): MQTT re-connect issue
Fixes #2197 .

- [x] This PR is for the `dev` branch rather than for `master`.
- [x] This PR is compliant with the [other contributing guidelines](https://github.com/nodemcu/nodemcu-firmware/blob/dev/CONTRIBUTING.md) as well (if not, please describe why).
- [x] I have thoroughly tested my contribution.
- [x] The code changes are reflected in the documentation at `docs/en/*`.

Connection with the broker could not be reestablished after a disconnect. This was caused by the node not sending a PINGREQ after the keep-alive period. This has been resolved by clearing the keep_al [... cut]