nodemcu/nodemcu-firmware #2199: enduser_setup ERROR: #3: 777  http_serve_html failed. tcp_write failed

enmotion, 2017-12-19:
I have a problem about  enduser_setup.start();
when I run it I got this  print information back
------------------------------------------------------------------------------------------------
E:M 1480
enduser_setup: Err #3: 777: 	http_serve_html failed. tcp_write failed
enduser_setup: Err #3: 1164: 	http_recvcb failed. Unable to send HTML
------------------------------------------------------------------------------------------------
if I am doing nothing after this information showing, keep waiting and do not setup wifi information
it goes to showing again and again untill  to dead and restart clips again
I don't konw how to fix the problem  some one can help me?
thanx alot

8<------------------------ here is code -----------------------------------------
WM.uesDly=tmr.create();
 wifi.sta.clearconfig()
    wifi.setmode(wifi.STATIONAP)
    WM.uesDly:register(1000,tmr.ALARM_SINGLE, function()
        print('start setup')        
        enduser_setup.manual(true)
        enduser_setup.start(
          function()end,
          function(err, str)
            print("enduser_setup: Err #" .. err .. ": " .. str)
          end
        );
        wifi.ap.config({ssid="MoreZ_"..string.sub(WM.devId,-6,-1), pwd="zerom"..string.sub(WM.devId,-4,-1)})          
    end)
    WM.uesDly:start();
8<--------------------------------------------------------------------------------
my module are
adc,bit,dht,encoder,enduser_setup,file,gpio,http,i2c,mqtt,net,node,ow,pcm,pwm,rfswitch,si7021,sjson,spi,tmr,uart,wifi

marcelstoer, 2017-12-21:
I doubt this is a bug with the module so I'm inclined to refer to #1010 but I'm not 100% sure.

Source: https://github.com/nodemcu/nodemcu-firmware/blob/master/app/modules/enduser_setup.c#L774
Note that Lua doesn't use `;` to mark statements.

closed 2018-02-04 by marcelstoer

cassc, 2020-03-16:
@haicanhdieu Hi, I tried dev branch build from https://nodemcu-build.com/, it works.