nodemcu/nodemcu-firmware #2200: Sending email using smtp.gmail.com. Only empty messages o'k.

av-user, 2017-12-19:
Firmware used:

NodeMCU custom build by frightanic.com
	branch: master
	commit: 443e8219527f5b2190324a969a4586f9d3d731bf
	SSL: true
	modules: file,gpio,net,node,tmr,uart,wifi,tls
 build 	built on: 2017-12-17 13:20
 powered by Lua 5.1.4 on SDK 2.1.0(116b762)

8<------------------------ BUG REPORT -----------------------------------------
### Expected behavior
sent: DATA
response: 354  Go ahead
sent: blablabla
sent: .
response: 250 2.0.0 OK
### Actual behavior
sent: DATA
response: 354  Go ahead
sent: blablabla
sent: .
NO RESPONSE from server. The server does not recognise the end of data. Only if "\r\n.\r\n" follows "DATA" command immediately (empty mail), the server responds with 250 2.0.0 OK
### Test code
```Lua
local smtp_socket = tls.createConnection()
...
smtp_socket:on("connection",connected)
smtp_socket:on("receive", response_hdl)
smtp_socket:connect(465, "smtp.gmail.com")
```
### NodeMCU version
see above, at the beginning

### Hardware
something like this:
https://user-images.githubusercontent.com/19575299/34166631-aaf4fa08-e4df-11e7-8d63-9b2e499ab0d0.png
8<------------------------ END BUG REPORT -------------------------------------

pjsg, 2017-12-19:
What is your actual test code. The message from gmail indicates that it did not think that you sent \r\n.\r\n to terminate your message. 

Are you sure that the string 'data' that you sent was terminated with \r\n?

av-user, 2017-12-19:
Hi! Thank you for reply.
My code you can find in this repository:
https://github.com/av-user/lua_node_mcu
, and here is a log I get sending "an empty message":

sent: DATA
state_data_sent response: 354  Go ahead v62sm16566494wrc.13 - gsmtp
response end
state_data_sent OK
sent: .
state_mail_sent response: 250 2.0.0 OK 1513704041 v62sm16566494wrc.13 - gsmtp
response end
state_mail_sent OK
sent: RSET
state_rset_sent response: 250 2.1.5 Flushed v62sm16566494wrc.13 - gsmtp
response end
state_rset_sent OK
sent: QUIT
state the end response: 221 2.0.0 closing connection v62sm16566494wrc.13 - gsmtp
response end
state_quit_sent OK

if I only try to send something (does not matter what) between "DATA" and "\r\n.\r\n", I get no response.

pjsg, 2017-12-19:
How do you do the multiple sends? You are supposed to wait for the previous send to complete before doing the next one -- otherwise you can run into cases where sends fail. Does

send("To: nobody@example.com\r\nFrom: me\r\n\r\nHello\r\n.")

work (in replace of your send(".")

av-user, 2017-12-19:
pjsg, thank you so much! I have find it in a few minutes, and I was struggling with this problem the last two days : ). I am new to lua (4 or 5 days).
It seems, that lua-socket behavior is not the same as of conventional one.
If there is a way to be notified when the socket is ready with sending, it would be the best way to fix is correctly.

av-user, 2017-12-19:
I found it! 

on(event, function())
Parameters
    event string, which can be "connection", "reconnection", "disconnection", "receive" or **"sent"**

jmcgnh, 2017-12-19:
I've seen several instances where fragmented transmissions (multiple write requests to a socket) work fine for http but fail for https, so the recommendation seems to be to prepare the entire transmission for sending in a single write call.

marcelstoer, 2017-12-21:
Good example: https://nodemcu.readthedocs.io/en/latest/en/modules/net/#netsocketsend

closed 2017-12-21 by marcelstoer