nodemcu/nodemcu-firmware #2206: HMC5883L settings support & changing data output formats

therealbattler, 2017-12-22:
### Missing feature
The HMC5883L library currently has no way of setting up the compass (features like number of samples averaged, data output rate, resolution and mode of operation). It also outputs the data that is not suited for proper heading calculation. The documentation is also not clear.

### Justification
Knowing how the chip is set up is crucial for its correct operation. Without knowing the resolution it is impossible to normalize the data and get an accurate heading readout, and setting up such things as mode of operation and mode of operation can also be useful. 

Moreover, currently the library outputs only the raw x,y,z data. It is hard to get accurate heading readout from the raw data as it is, but since the NodeMCU firmware does not have any trigonometric functions enabled, it is **impossible to calculate the heading on the chip itself** (an atan2 calculation is required). Without those features, the library becomes useless, as it does not do anything that can be achieved via a simple I2C readout.

Regarding the documentation, it provides little information about the readouts, for example:

> Returns
x,y,z measurements (integers) temperature multiplied with 10 (integer)

It is not specified whether the data is raw or not, the reason for multiplying by 10 is also not clear.

Overall, it is easier to not use the library whatsoever and communicate with the chip directly; that is why I think the library should be improved.

### Workarounds
The workaround is to communicate with the chip directly via the I2C bus without the library and perform the heading calculation on another device.

marcelstoer, 2017-12-22:
> I can contribute to this module myself and would be happy to do so.

Great, that's the spirit we need!

The module's author @jschmidlapp would be best to argue about the module's general usefulness I guess. Maybe your criticism applies to the author's other brother-in-mind modules ADXL345 and L3G4200D as well?

It's definitely valuable to have these modules even though the sensors they support serve "special" IoT use cases. No surprise neither of them is among the top-30 wrt installed-base as per https://nodemcu-build.com/stats.php.

edtamarin, 2017-12-22:
The module's narrow use case is why I posted here before actually diving deep into development. I will make some changes as I envisioned them during the next two weeks or so.

I haven't checked the other modules as I am working with the HMC at the moment, but I will take a look.

And of course, I will be happy to discuss this with @jschmidlapp

marcelstoer, 2017-12-22:
Might also want to take a look at #2188 before digging into code.

pjsg, 2017-12-22:
Suprisingly enough, floating point is available in the integer builds (but not as far as the LUA is concerned). It seems possible that atan2 is also available at the C level even in an integer build. yes, it will bring in a bunch of supporting code, but that maybe a worthwhile tradeoff.

devsaurus, 2017-12-23:
> floating point is available in the integer builds (but not as far as the LUA is concerned)

My understanding is that type `LUA_NUMBER` gets switched from `double` to `LUA_INTEGER`, that's all. So the functionality on module C level is not affected except for interfacing with Lua level.

@edtamarin when I looked into this (and other similar modules) I got the impression that the C implementation reliefs the user from programming I2C sequences on Lua level. That's not a lot of functional benefit, but worth enough to have it as a module IMO.
It would be great to sketch some specific use cases for this sensor and then draft the necessary API changes to support these cases.

edtamarin, 2017-12-23:
@devsaurus true, the module makes it so that the user can get the X,Y,Z axis data without directly accessing the I2C bus, but as I have written above, this is not very useful for the magnetometer, as the user would have to offload the heading calculation to another device.

The use cases I had in mind are:
- Easy heading readouts
- Access to the chip's settings for the user (resolution, sample averaging, etc.)
- Sensor calibration (accounting for the declination angle and maybe the sensor tilt)

The X,Y,Z data readouts can be left in the module.

closed 2019-08-04 by stale[bot]