nodemcu/nodemcu-firmware #2207: ads1115 is obfuscating a sign for values between 0 and -1mV in integer build

paweljasinski, 2017-12-23:
### Expected behavior
This is the case for integer build only
Positive and negative values derived from volt and volt_dec in the range -1mV .. +1mV should be different for different polarization. 

### Actual behavior
As the voltage raises above the -1mV, the volt is 0 and volt_dec is positive.

raising about -1 mV
------------------
volt volt_dec dac
-1	93	65466
-1	46	65469
0	984	65473
0	937	65476
0	875	65480

crossing zero
----------------
volt volt_dec dac
0	31	65534
0	15	65535
0	0	0
0	15	1
0	31	2

### Workaround
Look at the dac value and correct the sign.
or
Do yourself the dac conversion.
or
Use floating build

### Test code
```Lua
id, scl, sda = 0, 1, 2
i2c.setup(id, sda, scl, i2c.SLOW)
ads1115.setup(ads1115.ADDR_GND)
ads1115.setting(ads1115.GAIN_0_512V, ads1115.DR_8SPS, ads1115.DIFF_0_1, ads1115.CONTINUOUS)

main_timer = tmr.create()
main_timer:alarm(1000, tmr.ALARM_AUTO, function()
    volt, volt_dec, adc = ads1115.read()
    print(volt, volt_dec, adc)
end)
```
### NodeMCU version
dev branch, bee404c35890bbbcb4dd158991adb25621337a50

### Hardware
wemos D1 with ads1015 (max gain, differential between in0 and in1, address pin grounded, 3.3V power).

paweljasinski, 2018-01-15:
How about new set of specialized read functions.
read_float - returns mV float and raw, in case of int build error
read_raw - returns only raw
read_int - returns sign (-1,0,1), mV, uV and raw - this would be to mimic the existing one, 
but read_int could return just int32_t uV and raw

devsaurus, 2018-01-24:
Not sure what's the best solution, but I would avoid specialized read functions to keep the API lean.
Maybe something like

`read()` returns V, V_dec, raw and sign
- V is signed on float builds, unsigned on int builds.
- sign is `nil` on float and (-1,1) on int

The application can then adapt according to the value of sign:
```lua
if sign then
  -- int build
  print(string.format("%s%d.%03d", sign > 0 and "+" or "-", V, V_dec))
else
  -- float build
  -- just use V as it is
end
```

Guess we have the same issue with the ds18b20 module - missing sign between 0⁰C and -1⁰C?

paweljasinski, 2018-01-25:
I have some of ds18b20 around. I can test it as well.
If ds is also affected, should I create separate issue for ds, or can I piggybag here?

devsaurus, 2018-01-29:
> should I create separate issue for ds, or can I piggybag here?

I'd vote for a separate issue.

PR #2246 (2018-02-07): ads1115 - introduced sign into read and startread
* in float build, uV and sign are included in mV
* in int build, uV and mV are absolute, sign is -1, 0, 1
* added rounding of uV values
* added optional test function

Fixes #2207

- [x] This PR is for the `dev` branch rather than for `master`.
- [x] This PR is compliant with the [other contributing guidelines](https://github.com/nodemcu/nodemcu-firmware/blob/dev/CONTRIBUTING.md) as well (if not, please describe why).
- [x] I have thoroughly tested my contribution. Tests available under: https://github.com/paweljasinski/nodemcu-mispec/blob/mast [... cut]
closed 2018-02-07 by marcelstoer