nodemcu/nodemcu-firmware #2209: Provide SJSON streaming example for standard use case

marcelstoer, 2017-12-25:
I would argue that the most obvious streaming use case for SJSON is parsing a large JSON object retrieved from an online service provider (to extract just a few properties).

To make use of the streaming features you'd most likely have to use [`net.socket:hold()`](https://nodemcu.readthedocs.io/en/latest/en/modules/net/#netsockethold). I guess it would look similar to what @devsaurus used for [the PCM example](https://github.com/nodemcu/nodemcu-firmware/blob/dev/lua_examples/pcm/play_network.lua#L87).

Philip, do you happen to have such code samples ready?

pjsg, 2017-12-25:
I'll put up an example -- I don't use the `:hold` method -- I just process as the data arrives....

pjsg, 2017-12-26:
How about something like

```
-- Test sjson and github api

local s = tls.createConnection()
s:on("connection", function (sck, c)
  sck:send("GET /repos/nodemcu/nodemcu-firmware/git/trees/master HTTP/1.0\r\nUser-agent: nodemcu/0.1\r\nHost: api.github.com\r\nConnection: close\r\nAccept: application/json\r\n\r\n")
end)

function startswith(String,Start)
   return string.sub(String,1,string.len(Start))==Start
end

local seenBlank = false
local partial
local wantval = { tree=1, path=1, url=1 }
-- Make an sjson decoder that only keeps certain fields
local decoder = sjson.decoder({metatable=
    {__newindex=function(t,k,v) 
       if wantval[k] or type(k) == "number" then
         rawset(t,k,v)
       end
     end}})
local function handledata(s)
  decoder:write(s)
end
-- The receive callback is somewhat gnarly as it has to deal with find the end of the header
-- and having the newline sequence split across packets
s:on("receive", function (sck, c)
   if partial then
      c = partial .. c
      partial = nil
   end
   if seenBlank then
     handledata(c)
     return
   end
   while c do 
       if startswith(c, "\r\n") then
         seenBlank = true
         c = c:sub(3)
         handledata(c)
         return
       end
       local s, e = c:find("\r\n")
       if s then
         -- Throw away line
         c = c:sub(e+1)
       else
         partial = c
         c = nil
       end
   end
end)

local function getresult()
  local result = decoder:result()
  -- This gets the resulting decoded object with only the required fields
  print (result['tree'][4]['path'], "is at",
         result['tree'][4]['url'])
end
s:on("disconnection", getresult)
s:on("reconnection", getresult)

-- Make it all happen!
s:connect(443, "api.github.com")
```

marcelstoer, 2018-02-04:
Thanks Philip, I finally added it to the repo.

commit 555b5574 (2018-02-04): Add sjson streaming example

Fixes #2209.
closed 2018-02-04 by marcelstoer