nodemcu/nodemcu-firmware #2212: TLS Handshake failed

bayuaji732, 2017-12-27:
8<------------------------ BUG REPORT -----------------------------------------
### Expected behavior
I want to send dht dummy data to Raspberry middleware with TLS. so i can send secure data to Coap server and MQTT server at Raspberry Middleware.
### Actual behavior
dofile("CoAP_TLS.lua");
UDP server/client is set.
coap_create is called.
net_create is called.
TCP server/socket is set.
net_start is called.
TCP port is set: 443.
> state: 5 -> 0 (0)
rm 0
pm close 7
scandone

	wifi_event_monitor_handle_event_cb is called
Connecting...
Connecting...
scandone
state: 0 -> 2 (b0)
state: 2 -> 3 (0)
state: 3 -> 5 (10)
add 0
aid 3
cnt 

connected with MyPi, channel 6
dhcp client start...

	wifi_event_monitor_handle_event_cb is called
Connecting...
Connecting...
Connecting...
ip:192.168.0.17,mask:255.255.255.0,gw:192.168.0.1

	wifi_event_monitor_handle_event_cb is called
Connected to CoAP
  URL: coap://192.168.0.1:5683/r/home/kitchen
  Client ID: 1751566
pm open,type:2 0
net_create is called.
TCP server/socket is set.
net_on is called.
net_on is called.
net_start is called.
TCP port is set: 8080.
TCP ip is set: 192.168.0.1
socket_connect is called.
certificate 29d
client handshake start.
ip:192.168.0.17,mask:255.255.255.0,gw:192.168.0.1
net_create is called.
TCP server/socket is set.
net_on is called.
net_on is called.
net_start is called.
TCP port is set: 8080.
TCP ip is set: 192.168.0.1
socket_connect is called.
certificate 29d
client handshake start.

### Test code
### Lua
```Lua
conn=net.createConnection(net.TCP, 1)
net.cert.verify([[
-----BEGIN CERTIFICATE-----
MIICmTCCAgICCQDKAeGXqOnYizANBgkqhkiG9w0BAQsFADCBkDELMAkGA1UEBhMC
SUQxEjAQBgNVBAgMCUVhc3QgSmF2YTEPMA0GA1UEBwwGTWFsYW5nMREwDwYDVQQK
DAhNaWRkdGVjaDEQMA4GA1UECwwHTWlkZHNlYzESMBAGA1UEAwwJbG9jYWxob3N0
MSMwIQYJKoZIhvcNAQkBFhRiYXl1YWppNzMyQGdtYWlsLmNvbTAeFw0xNzEyMjEw
NzE2MjdaFw0xODEyMjEwNzE2MjdaMIGQMQswCQYDVQQGEwJJRDESMBAGA1UECAwJ
RWFzdCBKYXZhMQ8wDQYDVQQHDAZNYWxhbmcxETAPBgNVBAoMCE1pZGR0ZWNoMRAw
DgYDVQQLDAdNaWRkc2VjMRIwEAYDVQQDDAlsb2NhbGhvc3QxIzAhBgkqhkiG9w0B
CQEWFGJheXVhamk3MzJAZ21haWwuY29tMIGfMA0GCSqGSIb3DQEBAQUAA4GNADCB
iQKBgQDPt2KHW2O26/bpM9uK2yzchSLLJ9X4ssQjapkt/9rDUOinFkMfu5ttq568
0g0q9aIbTU61Hpoceu2IzAFXUSeEkBS8tT6cldxPAm27gp93mOxiDkyHBmxVRCRR
zyK4McLQety3E8mdXNVery4PsnJOdYOCuUuIHO9IITRLz+YmWQIDAQABMA0GCSqG
SIb3DQEBCwUAA4GBAC2m3qxn8z6CyxrlLwmVwI9CXU+j4Ng95EVH7fhscSrxpJQM
9uurTJh0ZlaqEEOEujoEwBw9OCVBx6HR/2uUI5yBwAaYJhmZ4TA3gG//yesDAhnE
43BEBpqXaHT6S329bITMKWEzCfO2VpMlnKsDnYAnzKCEAYiDCHv4xgRL61J5
-----END CERTIFICATE-----
]])
net.cert.verify(true)
--http.get("https://www.howsmyssl.com/s/api.html", nil, function (code, resp) print(code, resp) end)

conn:on("connection",function(conn, payload)
    conn:send("HEAD / HTTP/1.1\r\n".."\r\n\r\n") 
    end)
conn:on("receive", function(conn, payload)
    time=string.sub(payload,string.find(payload,"Date: ")+6,string.find(payload,"Date: ")+34)
    --get_sensor_Data()
    cc:post(coap.CON, coap_url, cjson.encode({
		protocol = "coap",
        timestamp = time,
        topic = "home/kitchen",
        sensor = {
            tipe = "esp8266",
            index = coap_id,
            ip = ipaddr,
            module = "dht22"
        },
        humidity = {
            value = humidity,
            unit = "%"
        },    
        temperature = {
            value = temperature,
            unit = "celcius"
        }
    }))
    print("Going to deep sleep for "..(time_between_sensor_readings/1000).." seconds")
    end) 
    t = tmr.now()
    conn:connect(8080,'192.168.0.1')
end
```
### Server.js
```Server.js
(function () {
  let app, start, mqtt, coap, redis, ascoltatori, logger, argv, consign, server
  let setupAscoltatore, setup, configure, coapServer

  consign = require('consign')
  redis = require('redis')
  ascoltatori = require('ascoltatori')
  mqtt = require('mqtt')
  coap = require('coap')
  tls = require('tls')
  fs = require('fs')
  https = require('https')

 options = {
    key: fs.readFileSync('/home/pi/qoap-middleware/certs/server.key'),
    cert: fs.readFileSync('/home/pi/qoap-middleware/certs/server.crt'),
    requestCert: true,
    ca: fs.readFileSync('/home/pi/qoap-middleware/certs/ca.pem'),
    rejectUnauthorized: false
}

  module.exports.app = app = require('express')()
  server = require('http').Server(app)
  
  app.io = require('socket.io')(server)
  app.redis = {}

module.exports.setupAscoltatore = setupAscoltatore = (opts) => {
    if (opts == null) {
      opts = {}
    }
    app.ascoltatore = new ascoltatori.RedisAscoltatore({
      redis: redis,
      port: opts.port,
      host: opts.host,
      db: opts.db
    })
    return app.ascoltatore
  }

  module.exports.setup = setup = (opts) => {
    let args
    if (opts == null) {
      opts = {}
  }
    args = [opts.port, opts.host]
    app.redis.client = redis.createClient.apply(redis, args)
    app.redis.client.select(opts.db || 0)
    return setupAscoltatore(opts)
  }

  module.exports.configure = configure = () => {
    return consign({cwd: 'lib', verbose: false})
      .include('models')
      .include('helpers')
      .include('controllers')
      .into(app)
  }

  start = module.exports.start = (opts, cb) => {
    configure()
    logger = app.helpers.winston
    argv = app.helpers.yargs

 if (opts == null) {
      opts = {}
    }
    if (cb == null) {
      cb = () => {}
    }

    opts.port || (opts.port = argv['port'])
    opts.coap || (opts.coap = argv['coap'])
    opts.mqtt || (opts.mqtt = argv['mqtt'])
    opts.mqttHost || (opts.mqttHost = argv['mqtt-host'])
    opts.redisPort || (opts.redisPort = argv['redis-port'])
    opts.redisHost || (opts.redisHost = argv['redis-host'])
    opts.redisDB || (opts.redisDB = argv['redis-db'])

    if (argv['ipv6']) opts.ipv6 = '::'

    setup({
      port: opts.redisPort,
      host: opts.redisHost,
      db: opts.redisDB
    })

    server.listen(opts.port, () => {
      logger.socket('Websocket listening on port %d in %s mode', opts.port, pro$
      logger.http('HTTP server listening on port %d in %s mode', opts.port, pro$
    })

    coapServer = coap.createServer(options)
    coapServer.on('request', app.controllers.coap_api).listen(opts.coap, opts.i$
      logger.coap('CoAP server listening on port %d in %s mode', opts.coap, pro$
    })

    new mqtt.Server(app.controllers.mqtt_api).listen(opts.mqtt, opts.ipv6, () =$
      logger.mqtt('MQTT server listening on port %d in %s mode', opts.mqtt, pro$
    })

    return app
  }

  if (require.main.filename === __filename) {
    start()
  }
}).call(this)
```
### NodeMCU version
NodeMCU custom build by frightanic.com
	branch: 1.5.4.1-final
	commit: 1885a30bd99aec338479aaed77c992dfd97fa8e2
	SSL: true
	modules: cjson,coap,crypto,dht,file,gpio,http,mqtt,net,node,tmr,uart,wifi,tls
 build 	built on: 2017-12-27 10:08
 powered by Lua 5.1.4 on SDK 1.5.4.1(39cb9a32)

### Hardware
ESP8266

8<------------------------ END BUG REPORT -------------------------------------

marcelstoer, 2017-12-27:
The 1.5.4.1-final branch is frozen and won't be changed anymore. You'll want to test against current master/dev versions.

closed 2017-12-27 by marcelstoer

bayuaji732, 2017-12-28:
thanks for your reply @marcelstoer 
now i try use

NodeMCU custom build by frightanic.com
	branch: dev
	commit: 05b0a0134ce42600ccfe8b799eafe9bd338eec63
	SSL: true
	modules: coap,crypto,dht,file,gpio,http,mqtt,net,node,sjson,tmr,uart,wifi,tls
 build 	built on: 2017-12-27 12:10
 powered by Lua 5.1.4 on SDK 2.1.0(116b762)

but still got same issue:

connected with MyPi, channel 6
dhcp client start...

	wifi_event_monitor_handle_event_cb is called
Connecting...
Connecting...
Connecting...
ip:192.168.0.140,mask:255.255.255.0,gw:192.168.10.2

	wifi_event_monitor_handle_event_cb is called
Connected to CoAP
  URL: coap://192.168.0.1:5683/r/home/kitchen
  Client ID: 1751566
pm open,type:2 0
client handshake start.
espconn_mbedtls.c 652, type[certificate],length[669]
please start sntp first !
client handshake failed!
Reason:[-0x7880]

could you tell me what i need to do? thanks

marcelstoer, 2017-12-28:
All those mbedTLS error codes are "documented" in `ssl.h`; yours is here: https://github.com/nodemcu/nodemcu-firmware/blob/master/app/include/mbedtls/ssl.h#L77. For whatever reason the peer hung up on you.

In November over on #1707 @nwf [mentioned the possibility of him carving out a PR](https://github.com/nodemcu/nodemcu-firmware/issues/1707#issuecomment-345129188) that would address _some_ TLS issues. You might want to give his dev-ssl branch a try (need to [build it with Docker](https://nodemcu.readthedocs.io/en/latest/en/build/#docker-image)).