nodemcu/nodemcu-firmware #2213: Stop supporting multi-language docs

marcelstoer, 2017-12-28:
Lets face it, nobody is ever going to translate our docs from English into something else. Close to 80 files, 65k words on 13k lines...no way.

So, this hint about ["outdated and sparse documentation"](https://nodemcu.readthedocs.io/en/latest/#outdated-and-sparse-documentation) is realistically speaking very obsolete. We should also remove the language subdirectory below `/docs`. This would allows us to dump 80% of [the JavaScript trickery](https://github.com/nodemcu/nodemcu-firmware/blob/dev/docs/js/extra.js) I wrote two years ago. To remain search engine friendly and avoid dead links we can set up [user-defined redirects in RTD](https://docs.readthedocs.io/en/latest/user-defined-redirects.html).

Objections?

marcelstoer, 2018-01-08:
Update: RTD redirects don't offer what we'd need (yet) but there are workarounds.

-> rtfd/readthedocs.org#3465

MacSass, 2018-01-08:
Hi, don´t think my opinion is worth a lot, but I don´t see a huge benefit in having multi language docs - I don´t think it will extend the user base to a large extent.
With the improvements in online translations multi language docs will be moot soon anyhow (check the quality of a pasted doc into deepl.com - it is more than readable).
Regards - MacSass

hanfengcan, 2018-02-21:
**if anyone need chinese doc, maybe read my article in [here](https://www.jianshu.com/nb/7000517)**

marcelstoer, 2018-08-21:
By coincidence I just found https://readthedocs.org/projects/nodemcu-cn/. @nodemcu is that you?

marcelstoer, 2019-01-12:
It's now possible to achieve this with a single global redirect rule as per https://github.com/rtfd/readthedocs.org/issues/3465: `/en/latest/en/$rest -> /en/latest/`. As this affects all branches I just need to do the restructuring in this repository for all branches at the same time (immediately cherry-pick from `dev` to `master`).