nodemcu/nodemcu-firmware #2216: `3ffef6c8 already freed` error on using `gpio.serout` in synchronous mode

Xasin, 2018-01-02:
Hello NodeMCU dev team,
Today I wanted to debug my code using my new logic analyzer, by inserting a few `gpio.serout()` commands at different points in the code and measuring the timing between them.

### Expected behaviour
The ESP-01 should have toggled pin 4 (`GPIO02`) on for 50us, off for 25, using the following line:
`gpio.serout(4, 1, {50, 25})`

### Actual behaviour
The ESP-01 _did_ toggle the pin with correct timing.
However, from the second use of the command onwards it additionally also throws the debug message `3ffef6c8 already freed`
This error message is thrown before the pin starts toggling, and does not change between restarts of the ESP. 
It seems to happen on the _next_ use of `gpio.serout` (synchronous and asynchronous) after a synchronous `gpio.serout`.
The ESP resumes normal function after this, it does not crash; however, the debug message interferes with the communication to an AVR and is undesirable.

### Test code
The following code was tested after a filesystem format to ensure no `init.lua` was present.
```Lua
gpio.mode(4, gpio.OUTPUT);

-- No error message thrown
gpio.serout(4, 1, {50, 25}, 1, nil);

-- Error thrown in synchronous mode
gpio.serout(4, 1, {50, 25}, 1, nil);

-- Error also thrown in asynch mode, and only once even for multiple iterations
gpio.serout(4, 1, {50, 25}, 2, 1);

-- Error not thrown after an asynchronous serout
gpio.serout(4, 1, {50, 25}, 1, 1);
gpio.serout(4, 1, {50, 25}, 1, nil);
```
### NodeMCU version
```
NodeMCU custom build by frightanic.com
	branch: master
	commit: 5073c199c01d4d7bbbcd0ae1f761ecc4687f7217
	SSL: false
	modules: adc, encoder, file, gpio, mqtt, net, node, rtctime, sjson, sntp, tmr, uart, wifi
        built on: 2018-01-02 09:25
 powered by Lua 5.1.4 on SDK 2.1.0(116b762)
```

### Hardware
The used ESP module was a generic ESP-01. 4MB of flash, and at the time of testing the above code, `node.heap()` returned `44488`
No outside hardware is required to reproduce the problem. 
Using a logic analyzer reveals that the error message is thrown before the pin toggling begins.

devsaurus, 2018-01-04:
I can reproduce this issue. Sounds like a bug in the heap memory usage by the serout feature.

devsaurus, 2018-01-04:
@Xasin the referenced PR fixes the error message for me. Thanks for reporting!

Xasin, 2018-01-04:
No problem, @devsaurus!
Aside from this little problem, NodeMCU has been running wonderfully for me, so I am happy to contribute!

Good to know the issue is taken care of.

PR #2220 (2018-01-10): Fix gpio.serout heap management
Fixes #2216.

- [x] This PR is for the `dev` branch rather than for `master`.
- [x] This PR is compliant with the [other contributing guidelines](https://github.com/nodemcu/nodemcu-firmware/blob/dev/CONTRIBUTING.md) as well (if not, please describe why).
- [x] I have thoroughly tested my contribution.

`lgpio_serout()` was missing to set `serout.delay_table` to NULL at the end of a synchronous operation. This triggered the `3ffef6c8 already freed` error message for the subsequent operation as reported in #2216.
closed 2018-01-10 by marcelstoer