nodemcu/nodemcu-firmware #2218: static ip, but getting also from dhcp server another ip

renegat56, 2018-01-03:
I have a lot of esp-01 running as web client with static ip. So they should not request an ip from dhcp server (ics-dhcp-server on linux machine). But they do. The client are only short time on to exchange data with the webserver (apache2 on linux machine), after that going to deep sleep. After every awake I get an entry in my dhcp.leases file, which will let it grow rapidly. Example:

lease 192.168.1.228 {
  starts 3 2018/01/03 10:01:51;
  ends 6 2018/01/13 10:01:51;
  tstp 6 2018/01/13 10:01:51;
  cltt 3 2018/01/03 10:01:51;
  binding state active;
  next binding state free;
  rewind binding state free;
  hardware ethernet 5c:cf:7f:80:b0:8e;
  client-hostname "NODE-80B08E";
}

Second issue : even I gave the esp-01 a new hostname, it is still named "NODE-80B08E" in the dhcp-leases file

### Test code
Provide a [Minimal, Complete, and Verifiable example](http://stackoverflow.com/help/mcve) which will reproduce the problem.
```Lua
wifi.sta.sethostname("Momo")
wifi.setmode(wifi.STATION)
wifi.sta.config {ssid="XXXX",pwd="YYYYY"}
wifi.sta.setip({ip="192.168.1.92",netmask="255.255.255.0",gateway="192.168.1.99"})
print("Current hostname is: \""..wifi.sta.gethostname().."\"")
wifi.sta.connect()
```
### NodeMCU version
NodeMCU custom build by frightanic.com
	branch: master
	commit: 2e67ff5a639a13260fd4fb3c3b627ccdc2845616
	SSL: false
	modules: adc,ads1115,bit,dht,ds18b20,file,gpio,i2c,net,node,ow,pcm,rtcmem,tmr,uart,wifi
built on: 2017-08-26 21:04
powered by Lua 5.1.4 on SDK 2.1.0(116b762)

### Hardware
ESP-01

devsaurus, 2018-01-04:
You're maybe fighting against wifi connecting automatically to the AP which is enabled by default.
Does it improve when you add `auto=false` to [`wifi.sta.config()`](http://nodemcu.readthedocs.io/en/dev/en/modules/wifi/#wifistaconfig)?

renegat56, 2018-01-06:
@devsaurus 
adding auto=false to wifi.stat.config() really helped. At least it does not appear in dhcp.leases anymore.
Thanks!

closed 2018-01-06 by renegat56