nodemcu/nodemcu-firmware #2219: Cannot connect to wi-fi network from NodeMCU board

KonradKlimczak, 2018-01-04:
I am trying to connect to my wifi network on my NodeMCU board. Not sure if it's a hardware or software problem, but I could not find any help on the issue.

I am trying to use this code for connecting to the WiFi:

    wifi.setmode(wifi.STATION)
    
    station_cfg={};
    station_cfg.ssid="netia9000";
    station_cfg.pwd="mywifipassword";
    wifi.sta.config(station_cfg)
    wifi.sta.connect()
    status_of_wifi = wifi.sta.status()
    
    if status_of_wifi == wifi.STA_IDLE then print("IDLE") end;
    if status_of_wifi == wifi.STA_CONNECTING then print("CONNECTING") end;
    if status_of_wifi == wifi.STA_WRONGPWD then print("WRONG PS") end;
    if status_of_wifi == wifi.STA_APNOTFOUND then print("404") end;
    if status_of_wifi == wifi.STA_FAIL then print("500") end;
    if status_of_wifi == wifi.STA_GOTIP then print("IP GOT") end;
    
    print(wifi.sta.getip())

But on console I can read following:

    CONNECTING
    nil

I tried to put the wrong data - a WiFi SSID that does not exist, a wrong `password, but no matter what I am still getting the same output: "CONNECTING" and "nil".

I used this code to check for available networks:

    wifi.setmode(wifi.STATION)
    
    -- print ap list
    function listap(t)
          for ssid,v in pairs(t) do
            authmode, rssi, bssid, channel = 
              string.match(v, "(%d),(-?%d+),(%x%x:%x%x:%x%x:%x%x:%x%x:%x%x),(%d+)")
            print(ssid,authmode,rssi,bssid,channel)
          end
    end
          
    wifi.sta.getap(listap)

And this worked perfectly fine. I got on the console:

    netia9000       3       -52         e8:11:23:43:bf:a2:8f        10
    -- other wi fi networks available nearby --

So it looks like the wifi module is fine and it's a software problem. I wrote the code according to [the documentation][1]. At this point I have no idea what's wrong. Any suggestions?

  [1]: https://nodemcu.readthedocs.io/en/master/en/modules/wifi/#wifistaconfig

devsaurus, 2018-01-04:
You probably expect that `wifi.sta.connet()` blocks and returns after connection is done. That would be a wrong assumption - it comes back immediately while the connection process is running in the background.
Use the [`wifi.eventmon`](https://nodemcu.readthedocs.io/en/master/en/modules/wifi/#wifieventmon-module) module instead to obtain events like `wifi.eventmon.STA_CONNECTED` from the wifi layer.

Much of the event-driven aspects are covered in the FAQ at [ESP8266 specifics](https://nodemcu.readthedocs.io/en/master/en/lua-developer-faq/#esp8266-specifics). It's a good read to get accustomed with the firmware principles.

marcelstoer, 2018-01-05:
Handled here: https://stackoverflow.com/q/48100864/131929

closed 2018-01-05 by marcelstoer