nodemcu/nodemcu-firmware #2223: support for ads1015

paweljasinski, 2018-01-07:
### Missing feature
When using ads1015 chip, the existing ads1115 module can be used. However, not everything maps 1:1.

- The single shot mode module programs an esp timer with the values which are incorrect. The interval values are longer than necessary (e.g. 10ms is programmed  where 1ms would be sufficient)

- Interpretation of constant values used to program sampling frequency is different (e.g. ads1115.DR_8SPS is equivalent of ads1015.DR_128SPS)

### Justification
ads1015 breakout boards are is available from adafruiit and ali

### Workarounds

- ads1115module can be used in continuous mode

- use ads1115 module as-is with known limitations

marcelstoer, 2018-01-08:
#2188 is similar in that it adds support to a _variant_ of a chip that we already have a module for. Not sure if it helps to look for inspiration there.

paweljasinski, 2018-01-09:
Thanks for the tip. I am new here, so for me it is not obvious how far is the memory consumption an issue.
We could make the type of the chip a parameter of the initialization. This will increase size of the module. Mostly parameter validation and different timer values.
Alternatively, a chip type could be a build time parameter - the module is compiled either 1015 or 1115.
Another, perhaps exotic requirements, is use of mixed setup. 1015 and 1115 in the same project. But before this, the module should support more than one chip - I don't believe the module in current state supports conveniently more than than one device.

Looking at the chip specs, I see no way to automate the initialization - from i2c side both chips appear identical.
Actually when I think more about it, I could configure "conversion ready mode" and measure how long did it take. Unfortunately, this is only if "conversion pin ready" is wired.

paweljasinski, 2018-01-15:
I manage to add support for multiple devices as well as different device type ads1015. I also believe it will work unchanged with ads1x13 and ads1x14. The changes can be seen under https://github.com/paweljasinski/nodemcu-firmware/tree/new-ads1115. I would appreciate feedback before updating documentation and doing a PR.

devsaurus, 2018-01-29:
PR merged. Thanks a lot for your contribution Pawel, very much appreciated!

PR #2231 (2018-01-29): added support for ads1015
Fixes #2223.

- [x] This PR is for the `dev` branch rather than for `master`.
- [x] This PR is compliant with the [other contributing guidelines](https://github.com/nodemcu/nodemcu-firmware/blob/dev/CONTRIBUTING.md) as well (if not, please describe why).
- [x] I have thoroughly tested my contribution.
- [x] The code changes are reflected in the documentation at `docs/en/*`.

This PR introduces support for ADS1015 based on the code for ADS1115.
In addition this PR includes:
* support for up to 4 converters (hardware limit).
* ads1115.setup() does not perform i2c genera [... cut]
closed 2018-01-29 by devsaurus