nodemcu/nodemcu-firmware #2224: Cant parse http response which is chunked

ugljanin, 2018-01-09:
I have script which coomunicates between nodemcu and my server. It  works good on my localhost and is parsing response retrieved from my  server when I send GET request. Problem is when I upload it all on my  website where transfer encoding is chunked. I am not able to retrieve  content, although request is legitimate and correct. Code is written in  Lua and I am trying to work on my NodeMCU device. 
` conn=net.createConnection(net.TCP, 0)     conn:on("connection",function(conn, payload)     conn:send("GET /mypath/node.php?id=1&update"..                 " HTTP/1.1\r\n"..                  "Host: www.mydomain.com\r\n"..                 "Accept: */*\r\n"..                 "User-Agent: Mozilla/4.0 (compatible; esp8266 Lua;)"..                 "\r\n\r\n")              end)      conn:on("receive", function(conn, payload)         if string.find(payload, "UPDATE")~=nil then              node.restart()         end          conn:close()         conn = nil      end)     conn:connect(80,"www.mydomain.com") end  `

Just to repeat that this GET request works and is tested manualy and  on localhost. Only problem is with chunked content, I don't know how to  parse it.

pjsg, 2018-01-09:
Try using HTTP/1.0 -- this doesn't support chunking (AFAIK), so it should be easier to parse.

ugljanin, 2018-01-09:
It removes chanked part, but still content is not displayed...

pjsg, 2018-01-09:
The code sample doesn't even try and display anything.....

ugljanin, 2018-01-09:
I have just created one file http.php with text included "php". I have uploaded it to localhost and to my domain. Once I tried to access my localhost from nodemcu, and then to domain. Results were different

This is the request

```
conn:send("GET /"..s.path.."/http.php"..
                " HTTP/1.0\r\n".. 
                "Host: "..s.domain.."\r\n"..
                "Accept: */*\r\n"..
                "User-Agent: Mozilla/4.0 (compatible; esp8266 Lua;)"..
                "\r\n\r\n") 
            end)
```
s.domain and s.path correcponds to different paths and domains on localhost and my domain

Result on domain
```
HTTP/1.1 200 OK
Date: Tue, 09 Jan 2018 03:09:28 GMT
Server: Apache
Connection: close
Content-Type: text/html; charset=UTF-8
```
result on localhost
```

TTP/1.1 200 OK
Date: Tue, 09 Jan 2018 03:08:48 GMT
Server: Apache/2.4.27 (Win64) PHP/7.0.23
X-Powered-By: PHP/7.0.23
Content-Length: 3
Connection: close
Content-Type: text/html; charset=UTF-8

php

```
As you can see, localhost is showing content, and domain is showing only header. When I type some file which does not exists domain is showing me html code.

pjsg, 2018-01-09:
You don't have the display code here. Note that the receive callback can be invoked one or more times with pieces of the HTTP response. It is a bad assumption that it all comes in one callback.

marcelstoer, 2018-01-09:
I see you track this at https://stackoverflow.com/q/48160461/131929. This ain't no user support forum.

closed 2018-01-09 by marcelstoer