nodemcu/nodemcu-firmware #2225: Upgrade to SDK 2.2

marcelstoer, 2018-01-10:
This is a placeholder issue for now. SDK 2.2 was announced for end of January 2018. However, we can use this issue to keep notes of "noteworthy" changes.

The long list of changes can be seen at https://github.com/espressif/ESP8266_NONOS_SDK/compare/v2.1.0...master

In light of #2214 (mbedTLS by @nwf) the following commits seem of particular interest i.e. should be checked for conflict during the upgrade. We don't want to loose our own fixes:
-  moved to new folder espressif/ESP8266_NONOS_SDK@89ba08e
-  Fix reconnect_callback is not triggered when handshake... espressif/ESP8266_NONOS_SDK@4899e50
-  speed up handshake process espressif/ESP8266_NONOS_SDK@493e8c0
-  ix disconnect callback function never be called espressif/ESP8266_NONOS_SDK@cf608e9
-  fix memory leak espressif/ESP8266_NONOS_SDK@a1fa90a
-  Fix call send callback function failed espressif/ESP8266_NONOS_SDK@5873d94

devsaurus, 2018-01-10:
Commit "fix(api): Fix ets_delay_us declaration" https://github.com/espressif/ESP8266_NONOS_SDK/commit/4671b17cc9fc6ed6787c2d310daf8accccf29c8d enables us to revert the changes I had to make in [`tmr.c`](https://github.com/nodemcu/nodemcu-firmware/commit/c4516f1a17a6aed75eec9aa778ad15a60560cdb7#diff-ac89d4cec75b9b260a2fe9e732d2c50bR131) and [`tsl2561.c`](https://github.com/nodemcu/nodemcu-firmware/commit/c4516f1a17a6aed75eec9aa778ad15a60560cdb7#diff-d5b868a3460dd619f0df04e15bfe5523R239) with https://github.com/nodemcu/nodemcu-firmware/commit/c4516f1a17a6aed75eec9aa778ad15a60560cdb7.

nwf, 2018-01-12:
A quick glance at the list @marcelstoer generated:
* espressif/ESP8266_NONOS_SDK@89ba08e does not seem relevant, given that we, too, have excised the pre-compiled mbedtls... unless there are plans to switch back to the SDK's and we believe that it will be kept up to date?
* espressif/ESP8266_NONOS_SDK@a1fa90a is a fix to LwIP; it probably should be applied to our tree, and won't conflict with any mbedTLS work.
* All the other commits are to third_party/mbedtls/app/espconn_mbedtls.c, the glue code between the SDK and mbedTLS itself; they seem like good things to port over and won't conflict with mbedTLS updates per se.

marcelstoer, 2018-02-06:
It landed: https://github.com/espressif/ESP8266_NONOS_SDK/releases/tag/v2.2.0

devsaurus, 2018-02-11:
I'd like to follow up on https://github.com/nodemcu/nodemcu-firmware/pull/2248#issuecomment-363497290 here since the other PR switched topic.

> The ones that just update *.a files are presumably of little importance to us, since we grab the SDK wholesale, including the .a files. Yes?

Correct, they're taken from the released SDK zip or removed during this process. I would even recommend to explicitly remove libssl.a as well.

> the thing to do is probably to extract the diffs from the sdk and mangle them to apply to nodemcu.

I agree with that but am lacking deeper git-fu to give recommendations on how to perform that in an efficient way. My best guess is to generate .patch files for each interesting commit with e.g.
`git format-patch -1 5ee8215c25e9b0ec1dcfcba13820022943c19158`, tailor each in the editor and re-apply them with something like `git am --sign-off < 0001-feat-espconn-Modification-for-espconn.patch`.

nwf, 2018-02-11:
@devsaurus That's more or less what I was doing, yes, though I generated the entire set of patches at once with `git format-patch -o patches v2.1.0`.   There might be a better way to apply the changes, but I don't know it.  There aren't *that* many commits of interest, tho', so this approach seems OK?

marcelstoer, 2018-02-22:
Maybe we can also make use of the new `version.h` as per https://github.com/espressif/ESP8266_NONOS_SDK/commit/05c812a1b1befa2bcd4ae861198e8b477eca4629

marcelstoer, 2018-03-08:
NodeMCU 2.2.0 will include the KRACK fix from Espressif.