nodemcu/nodemcu-firmware #2227: Possible memory leak in PCM module

redliquid, 2018-01-14:
### Expected behavior
The same memory amount available after destroying the object.

### Actual behavior
Free memory amount decreased by 16 bytes (not always, see https://pastebin.com/bprfxy3k)

### Test code
```Lua
function play()
  drv = pcm.new(pcm.SD, 5)
  drv = nil
  print("[" .. tmr.time() .. "], HEAP: " .. node.heap())
end

tmr.create():alarm(1000, tmr.ALARM_AUTO, play)
```
### NodeMCU version
This issue appears to be in `nodemcu:master` and `nodemcu:dev` branch.

Currently testing on:
NodeMCU custom build by frightanic.com
	branch: dev
	commit: df930c2d031bad8ab1d47ef4c06173d111e29e4f
	SSL: false
	modules: adc,bit,bme280,enduser_setup,file,gpio,http,i2c,net,node,pcm,sigma_delta,sjson,tmr,tsl2561,uart,wifi
 build 	built on: 2018-01-14 10:25
 powered by Lua 5.1.4 on SDK 2.1.0(116b762)

devsaurus, 2018-01-14:
Thanks for reporting! I'll have a look.

devsaurus, 2018-01-14:
@redliquid the leak highlighted by your test code gets fixed with the referenced PR .

PR #2228 (2018-01-15): Fix leak in pcm for subsequent object creation
Fixes #2227.

- [x] This PR is for the `dev` branch rather than for `master`.
- [x] This PR is compliant with the [other contributing guidelines](https://github.com/nodemcu/nodemcu-firmware/blob/dev/CONTRIBUTING.md) as well (if not, please describe why).
- [x] I have thoroughly tested my contribution.

This fixes a memory leak when multiple pcm objects are created:
No need to assign task ids in `new()`, just initialize them once during module registration.
closed 2018-01-15 by TerryE