nodemcu/nodemcu-firmware #2229: gpio.trig trigger wdt the nodemcu after node.restart()

crasu, 2018-01-15:
After a node.restart() gpio.trig() hangs and the wdt reset is triggered.

### Expected behavior
gpio.trig should not result in a wdt reset

### Actual behavior
It reboots twice

### Test code
I am using the following init.lua
```Lua
print("setting up mode for io index 11")
gpio.mode(11, gpio.INT, gpio.PULLUP)
print("setting trigger for io index 11")
gpio.trig(11, "down", function() end)
```

When I run node.restart this results in the following output:
```
> node.restart()
> 
 ets Jan  8 2013,rst cause:2, boot mode:(3,6)

load 0x40100000, len 26416, room 16 
tail 0
chksum 0x1d
load 0x3ffe8000, len 2176, room 8 
tail 8
chksum 0xfe
load 0x3ffe8880, len 136, room 0 
tail 8
chksum 0x88
csum 0x88
<GARBAGE REMOVED>
NodeMCU 2.1.0 build unspecified powered by Lua 5.1.4 on SDK 2.1.0(116b762)
setting up mode for io index 11
 <HANG FOR A COUPLE OF SECS>
 ets Jan  8 2013,rst cause:4, boot mode:(3,6)

wdt reset 
load 0x40100000, len 26416, room 16 
tail 0
chksum 0x1d
load 0x3ffe8000, len 2176, room 8 
tail 8
chksum 0xfe
load 0x3ffe8880, len 136, room 0 
tail 8
chksum 0x88
csum 0x88
<GARBAGE REMOVED>

NodeMCU 2.1.0 build unspecified powered by Lua 5.1.4 on SDK 2.1.0(116b762)
setting up mode for io index 11
setting trigger for io index 11
> 

### NodeMCU version
current master and dev branch from github

### Hardware
nodemcu devkit

### Analysis
The hang seems to happen in set_gpio_interrupt when reenabling interrupts.

devsaurus, 2018-01-18:
You're manipulating pin 11 (GPIO9) which is connected to the flash chip on many module variants (required for QIO flash mode). This jeopardizes code fetches and will most likely result in a hanging system.

Stay away from pin 11 and pin 12 (GPIO9 and GPIO10, respectively) unless you have a module that doesn't connect these to the flash chip.

crasu, 2018-01-18:
Thank you for your reply. This happens for other gpios as well. I just figured out that this does not happen always.

devsaurus, 2018-01-18:
Ok, I agree. That shouldn't cause a wdt reset.

With your sample code I can reproduce the following behavior:
* it appears to work 100% when executed after a hardware reset (even multiple times)
* it hangs 100% in the first iteration for `gpio.mode()` when executed after `node.restart()`

I'd guess that certain configuration settings applied by either `gpio.mode()` or `gpio.trig()` aren't cleared by `node.restart()`, while a hardware reset does clear them.
Nasty, let's see what I can dig out.

crasu, 2018-01-19:
This seems to fix it.
https://gist.github.com/crasu/b730915fb847addee4d5a3e987cc48b1

pjsg, 2018-01-19:
Actually, I think that the problem is in the interrupt handler. It appears that the setting of the mode configures the interrupt, but the pin_int_type doesn't get set until the trig method is called. During this time, if an interrupt happens, then it isn't handled correctly (especially if it is a level triggered interrupt). 

I *suspect* that the right thing to do is to disable the interrupt pin if pin_int_type is not set and an interrupt is taken.

In the short term, I think that the problem can be solved by switching the order of the mode and trig calls.

crasu, 2018-01-20:
Changing the order of trig and mode does not help.

I added some debug output to the dispatcher and set_gpio_interrupt
```
NodeMCU 2.1.0 build unspecified powered by Lua 5.1.4 on SDK 2.1.0(116b762)
setting up mode for io index 1
pin,mode,pullup= 1 2 1
Pin data at mode: 1 60000840, 5 0 0, fffffffe
Function platform_gpio_mode() is called. pin_mux:1610614848, func:0
set_gpio_interrupt reenabling interrupts now
platform_gpio_intr_dispatcher called
platform_gpio_intr_dispatcher called
platform_gpio_intr_dispatcher called
platform_gpio_intr_dispatcher called
platform_gpio_intr_dispatcher called
platform_gpio_intr_dispatcher called
....
```

PR #2234 (2018-02-13): Disable (and clear) unexpected interrupts
Fixes #2229.

- [x] This PR is for the `dev` branch rather than for `master`.
- [x] This PR is compliant with the [other contributing guidelines](https://github.com/nodemcu/nodemcu-firmware/blob/dev/CONTRIBUTING.md) as well (if not, please describe why).
- [ ] I have thoroughly tested my contribution.

As @pjsg suggested: When the GPIO ISR is triggered for an unexpected interrupt it's shut off and has its status cleared. This fixes the issue of pending interrupts after `node.restart()` and is supposed to mitigate races when a level-triggered interrupt  [... cut]
closed 2018-02-13 by marcelstoer