nodemcu/nodemcu-firmware #2230: New Feature: SM16703P support (similar to ws2812)

aux-lux, 2018-01-15:
### Missing feature
SM16703P support (similar to ws2812)

### Justification
General case: Those are (as far as I can tell) quite popular chinese alternatives for ws2812, yet they differ a bit from those with timing, which causes led strips to work incorrectly.

My case: I was looking so long to find white led strip with addressabel leds. I finally found ws2811 ones, yet after shipping it appeared to be SM16703P, and there are no alternatives.

### Hints
This is datasheet http://www.greeled.net/uploadfile/03/60/6003/21141/05/09/sm16703-english-datasheet.pdf
Hopefully setting timing to provided should work? I'm not sure if it's event possible.

pjsg, 2018-01-15:
Another possibility is that these chips are more sensitive to voltage levels... How are you driving the strip?

aux-lux, 2018-01-15:
I'm using 12v power supply for leds.
What I also found is that in here http://doityourselfchristmas.com/forums/showthread.php?44207-WS2811-LED-Strip-issue/page2 people seem to solve issue by changing only reset time from 50 to 80 (author later told me he even extended it to 300 for other strips support).
And that guys - https://www.youtube.com/watch?v=6zqGwxqJQnw - seems to fix this using mp3 library...
But I'm just simple FE developer. I love nodemcu since it uses lua :) But all that stuff above are way away from my current knowledge.

pjsg, 2018-01-15:
From looking at the code, it doesn't appear that there is a delay at all after sending the data. This could be fixed.

What does your driving lua code look like?

aux-lux, 2018-01-15:
I tried

```
ws2812.init()
ws2812.write(string.char(255, 0, 0, 255, 0, 0))
```
And this one (tu turn off at least one or two leds, since all are ON at start).
```
ws2812.init()
ws2812.write(string.char(0, 0, 0, 0, 0, 0))
```
```
ws2812.init()
local i, buffer = 0, ws2812.newBuffer(150, 3); buffer:fill(0, 0, 0, 0); tmr.create():alarm(50, 1, function()
  i = i + 1
  buffer:fade(2)
  buffer:set(i % buffer:size() + 1, 0, 0, 0, 255)
  ws2812.write(buffer)
end)
```
Last one did affected the way of displaying leds, but it was really chaotic.

aux-lux, 2018-01-17:
I'm afraid you sir might be right about sensitivity of this module. I tried to tinker a bit in ws2812 module and try to change os_delay_us(10) to 80 or even 300 witout success. I was also trying other baud rates, but I'm not sure if I managed to make any changes. I tried to change from 3200000 to 3333333. I was guessing, that it might be something like 1second / 3333333 = 0.3us. And I left os_delay_us(80);
Unfortunately without any success. But! After command:
```
ws2812.init()
ws2812.write(string.char(0, 0, 0, 0, 0, 0)) 
```
I've seen 2 first leds blink. Thats not much, and maybe was just random failure, but who knows.
Can someone confirm to me, that theoretically my change of baud should cause short signal least 0.3us and long signal 0.9us ? Or it's not working this way?

aux-lux, 2018-01-18:
I'm connecting led strip directly to pin 4 (GPIO2). I have only NPN bipolar transistor, and I have no idea, how I can use it in this case, and my mosfet is also manipulating ground wire. So I ordered PNP bipolar transistor and will let you know if it helped :)

pjsg, 2018-01-18:
The standard approach is to use something like https://www.sparkfun.com/products/12009 or similar from aliexpress

aux-lux, 2018-01-22:
I've got my logic level converter and it works I guess :) I used my build from last week, when I changed sleep to 80us, not tested on clean library yet where is 10us, but I will try later :) Sad thing is that my led strip works in thirds :( so I can only light up, dimm or turn off 3 leds at once. But with super high resolution :D Since each of 3 8bit fields are connected together making one 24bit brightness for 3 leds at once :P Guess it's because that chinese manufacturer hates people.

pjsg, 2018-02-13:
@aux-lux I'd like to close this ticket -- any objections?

aux-lux, 2018-02-22:
No problem, if I will have any update, then I will let you know ;)

NorthernMan54, 2018-10-09:
@pjsg @aux-lux I think you can close this, as I also have a SM16703 led strip ( from Costco ) and can report that it works correct without any changes to the nodemcu firmware ;-).

To make this work, I used the level convertor you mentioned earlier in the thread and wired everything up.  For the Costco strip I was using I initialized the buffer with 

strip_buffer = ws2812.newBuffer(24, 3)

Each strip section had 12 SM16703 chips on it.  It also has a separate white LED which is not wired to the SM16703, but is separately controlled.  And is designed to have a maximum of two strips connected together, so 24 LED's in total.

The Costco LED strip had 5 connections

24 Volt  - Wired to led strip power supply
DIN        - Wired via level convertor to pin D4 on my nodemcu
Check   - Left unconnected
W          - White led, connect to ground to turn on white LED
Gnd      - Connected to the led strip power supply ground and the nodemcu ground

DavijhonGMC, 2019-09-08:
> @pjsg @ aux-lux Acho que você pode fechar isso, pois também tenho uma faixa de led SM16703 (da Costco) e posso relatar que funciona corretamente sem alterações no firmware nodemcu ;-).
> 
> Para fazer isso funcionar, usei o conversor de nível que você mencionou anteriormente no thread e conecte tudo. Para a faixa Costco que eu estava usando, inicializei o buffer com
> 
> strip_buffer = ws2812.newBuffer (24, 3)
> 
> Cada seção de tira tinha 12 chips SM16703. Ele também possui um LED branco separado, que não é conectado ao SM16703, mas é controlado separadamente. E foi projetado para ter no máximo duas tiras conectadas, portanto, 24 LEDs no total.
> 
> A tira de LED Costco tinha 5 conexões
> 
> 
> Fonte de alimentação de 24 volts - com fio para tira DIN - Com fio através do conversor de nível para fixar o D4 no meu nodemcu
> Verifique - Esquerda desconectada - tem 24 volts durante a execução?
> W - Led branco, conecte ao terra para ligar o LED branco
> Gnd - Conectado ao terra da fonte de alimentação da tira de led e ao terra nodemcu
> 
> O padrão do LED é ímpar, e não o GRB normal, são RGB, que distorcem alguns dos efeitos do ws2812 e os fazem parecer desconectados.

----------
Desculpe, sou Brasileiro, e meu Inglês é ruim.
é possível direto no Raspberry pi 3? 
Ou o que é preciso fazer para controlar direto do Raspberry pi 3?

NorthernMan54, 2019-09-08:
```
Sorry, I'm Brazilian, and my English is bad.
Is it possible to control straight from Raspberry pi 3?
Or
What does it take to control straight on Raspberry pi 3?
```

@DavijhonGMC Sorry I'm not aware of a Raspberry PI implementation of this.   As an FYI, this is my final implementation of this work

https://github.com/NorthernMan54/homebridge-mculed

nwf, 2019-09-08:
This is not an appropriate forum to discuss Raspberry Pi hardware.

Can we close this ticket?

HHHartmann, 2019-09-09:
Closing since there were no objections since 1 year

closed 2019-09-09 by HHHartmann