nodemcu/nodemcu-firmware #2235: nodemcu: make flash :report error:  partitions-2MB.csv  Error at line 4: Value 'rf' is not valid. Known keywords: esphttpd, nvs, ota, coredump, fat, phy, spiffs

liuyang77886, 2018-01-22:
我按照这篇文章步骤：http://wiki.ai-thinker.com/esp32/examples/nodemcu
第一步：make menuconfig
第二步：修改串口为ttyS2，拷贝components/platform/partitions-2MB.csv到nodemcu-firmware 工程目录下，修改分区Partition Table.
 第三步：make flash 。 出现如下错误

LD build/bootloader/bootloader.elf
esptool.py v2.1
Building partitions from /cygdrive/g/esp/nodemcu/nodemcu-firmware/nodemcu-firmware-esp32/partitions-2MB.csv...
Error at line 4: Value 'rf' is not valid. Known keywords: esphttpd, nvs, ota, coredump, fat, phy, spiffs
make: *** [/cygdrive/g/esp/AiThinkerIDE_V0.5/cygwin/home/aithinker/project/esp32/esp-idf/components/partition_table/Makefile.projbuild:43：/cygdrive/g/esp/nodemcu/nodemcu-firmware/nodemcu-firmware-esp32/build/partitions-2MB.bin] 错误 2

Administrator@root-PC /cygdrive/g/esp/nodemcu/nodemcu-firmware/nodemcu-firmware-esp32
$

vowstar, 2018-01-23:
请更新esp-idf到最新的版本，首先cd到esp-idf所在目录，然后执行``git submodule update --init --recursive ``
确认esp-idf版本最新后，修改源码中你需要的配置，执行``make clean``, ``make defconfig``,`` make``, ``make flash``

marcelstoer, 2018-02-04:
As long as all is in Chinese you're unlikely to get help from the current maintainers - we don't understand what is requested here.

361790160, 2018-02-22:
me too,esp-idf is the latest and recursive

devsaurus, 2018-03-11:
> me too,esp-idf is the latest and recursive

Building with latest upstream esp-idf is not tested.
Last known-good build is with the referenced version of the esp32-esp-idf submodule:
```
$ cd sdk/esp32-esp-idf
$ git status
HEAD detached at 9a26296a
nothing to commit, working tree clean
```