nodemcu/nodemcu-firmware #2236: 使用官方NodeMCU-32S出厂固件，不能调用tmr，wifi等模块。

liuyang77886, 2018-01-22:
使用官方NodeMCU-32S出厂固件，不能调用tmr，wifi等模块。
根据资料http://wiki.ai-thinker.com/esp32/boards/nodemcu_32s?s[]=esplorer
使用NodeMCU-32S出厂固件ai-thinker_nodemcu-32s_dio_32mbit_20170705.7z，刷入芯片，使用ESPlorer编写lua脚本
编写print("hello") 写入芯片能正常运行。但是写入如下代码
tmr.alarm(1,5000,1,function() print("alarm 1") end)
tmr.alarm(1,5000,1,function() print("alarm 1") end)   
lua运行报错： attempt to call field 'alarm' (a nil value)
其他代码（wifi、pwm）也报错，代码是从http://www.nodemcu.com/index_cn.html#fr_5475f7667976d8501100000f里复制的

hanfengcan, 2018-01-23:
把代码保存成文件,save to esp 试试看行不行

相关的中文例程可看这里
https://www.jianshu.com/nb/7000517

liuyang77886, 2018-01-23:
我觉得是出厂固件没有tmr，wifi等模块，需要编译进去，但是云编译又没有32选项，用c的原厂固件有其他问题，所以才改用nodemcu

vowstar, 2018-01-23:
你可以尝试配置esp-idf环境（https://github.com/espressif/esp-idf），配置方法和操作系统相关：

* [Windows配置指南](https://esp-idf.readthedocs.io/en/latest/get-started/windows-setup.html)
* [Mac OS配置指南](https://esp-idf.readthedocs.io/en/latest/get-started/macos-setup.html)
* [Linux配置指南](https://esp-idf.readthedocs.io/en/latest/get-started/linux-setup.html)

然后使用esp-idf编译nodemcu esp32分支，编译前请启用你需要的功能，详细功能配置方法请仔细阅读github页面的文档。http://www.nodemcu.com首页的信息已经过时，请以github文档为基准。

marcelstoer, 2018-02-04:
As long as all is in Chinese you're unlikely to get help from the current maintainers - we don't understand what is requested here.

TerryE, 2018-07-21:
This is also more of a "how can I" rather than a "this is a firmware bug" anyway.