nodemcu/nodemcu-firmware #2237: nodeMCU固件云编译没有esp32的版本

liuyang77886, 2018-01-22:
nodeMCU固件云编译没有esp32的版本
网址：https://nodemcu-build.com/

hanfengcan, 2018-01-23:
你在这里发中文,不一定能解决问题吧

liuyang77886, 2018-01-23:
我都跟着做了的。以下为我所有的问题。
问题１：编写Pwm控制舵机程序，舵机无反应，但是led有闪烁 
根据资料http://wiki.ai-thinker.com/esp32从官网下载固件
git clone --recursive https://github.com/espressif/esp-idf.git
拷贝官方例子get-started的hello_world
编写Pwm控制舵机程序，1毫秒高电平19毫秒低电平，舵机转向为45度，舵机有单独供电5V3A电源，接线信号线（橙色）、5V电源正（红）、电源负（棕，与esp32共地）
程序文件testgpio.c，主要代码如下：
hw_gpio_init();
gpio_set_level(GPIO_BLUE_LED, 0);
        vTaskDelay(2000 / portTICK_RATE_MS); 
        gpio_set_level(GPIO_BLUE_LED, 1);
        vTaskDelay(1);//1毫秒高电平
        gpio_set_level(GPIO_BLUE_LED, 0);
        vTaskDelay(5000 / portTICK_RATE_MS);//5秒低电平

问题2：编写连接wifi程序，程序运行报错。
根据资料http://wiki.ai-thinker.com/esp32从官网下载固件
调试发现是在esp_wifi_init(&cfg)这里报错，我发现esp_wifi_init调用esp_wifi_init_internal，但是没有esp_wifi_init这个方法，只有在esp_wifi_internal.h里有定义没有实现。
程序文件testwifi.c，主要代码如下：
tcpip_adapter_init();
    ESP_ERROR_CHECK(esp_event_loop_init(event_handler, NULL));
    ESP_LOGI(TAG, "==========a");
    wifi_init_config_t cfg = WIFI_INIT_CONFIG_DEFAULT();
    ESP_LOGI(TAG, "==========b");
    ESP_ERROR_CHECK(esp_wifi_init(&cfg));
    ESP_LOGI(TAG, "==========c");
运行报错如下：
[0;32mI (22146) ========: ==========3[0m
[0;32mI (22146) ========: ==========4[0m
[0;32mI (22146) ========: ==========a[0m
[0;32mI (22146) ========: ==========b[0m
I (22146) wifi: wifi firmware version: 403db1d
I (22156) wifi: config NVS flash: enabled
I (22156) wifi: config nano formating: disabled
[0;32mIGuru Meditation Error: Core  1 panic'ed (LoadStoreError)
. Exception was unhandled.

问题3：我make flash经常报错，自己多试几次build project，make flash等操作就又成功执行make flash。
make[1]: Entering directory '/home/aithinker/project/hello_world/build/xtensa-debug-module'
make[1]: Nothing to be done for 'build'.
make[1]: Leaving directory '/home/aithinker/project/hello_world/build/xtensa-debug-module'
。。。。。。中间省略
python /home/aithinker/project/esp32/esp-idf/components/esptool_py/esptool/esptool.py --chip esp32 elf2image --flash_mode "dio" --flash_freq "40m" --flash_size "2MB"  -o /home/aithinker/project/hello_world/build/hello-world.bin /home/aithinker/project/hello_world/build/hello-world.elf
esptool.py v2.1
Flashing binaries to serial port /dev/ttyS2 (app at offset 0x10000)...
python /home/aithinker/project/esp32/esp-idf/components/esptool_py/esptool/esptool.py --chip esp32 --port "/dev/ttyS2" --baud 115200 --before "default_reset" --after "hard_reset" write_flash -z --flash_mode "dio" --flash_freq "40m" --flash_size detect 0x1000 /home/aithinker/project/hello_world/build/bootloader/bootloader.bin 0x10000 /home/aithinker/project/hello_world/build/hello-world.bin 0x8000 /home/aithinker/project/hello_world/build/partitions_singleapp.bin
esptool.py v2.1
Connecting....
Traceback (most recent call last):
  File "/home/aithinker/project/esp32/esp-idf/components/esptool_py/esptool/esptool.py", line 2524, in <module>
    _main()
  。。。。。。中间省略
  File "/home/aithinker/project/esp32/esp-idf/components/esptool_py/esptool/esptool.py", line 1559, in slip_reader
    waiting = port.inWaiting()
  File "/usr/lib/python2.7/site-packages/serial/serialutil.py", line 572, in inWaiting
    return self.in_waiting
  File "/usr/lib/python2.7/site-packages/serial/serialposix.py", line 456, in in_waiting
    s = fcntl.ioctl(self.fd, TIOCINQ, TIOCM_zero_str)
IOError: [Errno 22] Invalid argument
make: *** [/home/aithinker/project/esp32/esp-idf/components/esptool_py/Makefile.projbuild:55: flash] Error 1

问题4：nodemcu操作make flash到芯片报错
根据资料https://nodemcu.readthedocs.io/en/dev-esp32/从官网下载固件
git clone --branch dev-esp32 --recurse-submodules https://github.com/nodemcu/nodemcu-firmware.git nodemcu-firmware-esp32
根据资料http://wiki.ai-thinker.com/esp32/examples/nodemcu操作编译
执行make flash到芯片报错，如下：
partitions-2MB.csv Error at line 4: Value 'rf' is not valid. Known keywords: esphttpd, nvs, ota, coredump, fat, phy, spiffs
partitions-2MB.csv文件是官方代码里的分区表文件

问题5：使用官方NodeMCU-32S出厂固件，不能调用tmr，wifi等模块。
根据资料http://wiki.ai-thinker.com/esp32/boards/nodemcu_32s?s[]=esplorer
使用NodeMCU-32S出厂固件ai-thinker_nodemcu-32s_dio_32mbit_20170705.7z，刷入芯片，使用ESPlorer编写lua脚本
编写print("hello") 写入芯片能正常运行。但是写入如下代码
tmr.alarm(1,5000,1,function() print("alarm 1") end)
tmr.alarm(1,5000,1,function() print("alarm 1") end)   
lua运行报错： attempt to call field 'alarm' (a nil value)
其他代码（wifi、pwm）也报错，代码是从http://www.nodemcu.com/index_cn.html#fr_5475f7667976d8501100000f里复制的

问题6 nodeMCU固件云编译没有esp32的版本
网址：https://nodemcu-build.com/

问题7： esp32芯片上有两个按钮，左边一个按钮旁有“en”字样， 右边一个是“i00”字样，是做什么的呢

vowstar, 2018-01-23:
请先确认脚本中的串口号是否正确，如果是Linux系统会容易一些，直接把``/dev/ttyS2``改成你对应的设备（例如``ttyUSB0``），如果是windows系统，并且你在cygwin或mysys中，那么``COM3``对应``/dev/ttyS2``，``COM4``对应``/dev/ttyS3``，以此类推。建议使用Linux系统进行开发，会容易理解一些。

marcelstoer, 2018-02-04:
As long as all is in Chinese you're unlikely to get help from the current maintainers - we don't understand what is requested here.

TerryE, 2018-03-02:
But google translate is enough to know that this is more of a howto problem, and not an underlying issue being reported.

closed 2018-03-02 by TerryE