nodemcu/nodemcu-firmware #2238: ads1115.setup() performs reset of all i2c devices

paweljasinski, 2018-01-22:
### Missing feature
The current released ads1115 module issues a i2c general reset during the setup()
This reset is not specific to ads hardware.
How about moving it to the i2c module.

### Justification
It is likely another module using i2c implements the same reset logic.

### Workarounds
The setup of ads1115 has to be called before other i2c devices are programmed.

devsaurus, 2018-01-24:
I'm not familiar with the ads1115 architecture, but I see that you moved the I2C reset command into a dedicated function with #2231 - is this issue here still applicable once #2231 got merged?

paweljasinski, 2018-01-25:
The ads1115 specific issue is gone, but in my opinion, the reset function does not belong to ads1115 module.
For example: In addiction to ads1115 I have dac on the i2c bus, and it does obey the general reset command. The reset command is not specific to ads1115. It feels natural to move the reset to i2c.

devsaurus, 2018-02-04:
> the reset function does not belong to ads1115 module.

After reading through the [I<sup>2</sup>C spec](https://www.nxp.com/docs/en/user-guide/UM10204.pdf) chapters for "General call address" and "Software reset" I also think that the generic bus reset should not be part of a device-specific module.

Consequently, the `i2c` module might benefit from a function to issue general calls. That would be more for user's convenience than driven by functional requirements, since the corresponding sequence can be generated with existing `i2c` functions. Something like
```lua
i2c.general_call(id, second_byte[, <more bytes>])
-- with constants for <second_byte>
-- i2c.RESET_AND_WRITE_SLA
-- i2c.WRITE_SLA
```

Is this a reasonable improvement of the `i2c` module? Any objections from others?

devsaurus, 2019-07-21:
Issue should be kept.

nwf, 2020-07-18:
Probably still an issue.  Belatedly, I see no reason not to add `general_call` functionality to the `i2c` module.

HHHartmann, 2021-07-21:
Still hoping to get this fixed