nodemcu/nodemcu-firmware #2239: sigma-delta signal irregularities

RadianM, 2018-01-24:
Although  "sigma-delta" signal generator module was added a while back #1000 I'm finding various signal corruptions in all branches most notably when used with sigma_delta.setpwmduty()

Expected behavior: setting a static value e.g. sigma_delta.setpwmduty(127) should produce a steady 127/128 ratio square wave e.g. 1.5875us low, 1.6125us high.

Actual behaviour is that an extended low pulse of 3.2us occurs every 200us when viewed on an oscilloscope.

To reproduce connect pin 1 to an audio amplifier via the RC filter shown in http://nodemcu-firmware.readthedocs.io/en/dev/en/modules/pcm/ and issue
```
sigma_delta.setup(1)
sigma_delta.setpwmduty(127)
```
This leads to a clearly audible 5KHz tone when using a filter to remove the 312.5KHz PWM period in my setup using a ESP8266-12 device. All other PWM values exhibit similar extended pulses (values <128 extend low, >128 extend high) except for 128 which produces a steady 1.6us low/ 1.6us high. 128 is the only value that produces a glitch free output.

However, as PWM values decrease/increase about 128, the frequency of extended pulses increases so pushing their effect on the filtered signal out of the audio band. At PWM value132 for example the extended pulse happens every 50us (20KHz). Unfortunately when used for audio reproduction the spurious tones are most prominent when mostly silence is required.

I stumbled upon this while investigating the adaptation of the PCM module to facilitate fixed-frequency PWM in #1255.

pjsg, 2018-01-24:
I think that this may be a report of the same problem: http://bbs.espressif.com/viewtopic.php?t=2253

RadianM, 2018-01-25:
Yes that's it. Not much follow-through from there. There still doesn't seem to be much in the way of desription of the delta-sigma module in the soc. Maybe Arnim chose to use variable frequency delta-sigma in the PCM module because the artifacts are pushed out to 32KHz? I think it's probably safe to assume that this is a silicon limitation/bug given that the only access we have is via the GPIO_SIGMA_DELTA register at 0x60000368

devsaurus, 2018-01-27:
I agree that we can't do anything in software without further info from Espressif.

RadianM, 2018-01-28:
Is it known if this hardware bug is also present in the ESP32?

pjsg, 2018-01-28:
I'm wondering if these pulses can be eliminated by different choices of `prescale` and `target`. Yes, the frequency will change somewhat, but this might be a better outcome.

RadianM, 2018-01-29:
I don't think that can be made to work pjsg. Any given `target` sets the duty factor regardless of the value of `prescale`. While counting through all possible prescale values for a given target DF I see no magic values that prevent the missing pulses. So it's various values of DF that are 'poisoned'.

I can't quite get my head around how much poisoning is going on though: even values are affected less than odd ones and exact 2^n values are seemingly OK so it's all a bit crazy.

I would say therefore that the Sigma Delta module is accurate for generating simple waveforms with duty factors of 1/2, 1/4, 1/8, 1/64 etc. and is the only high-frequency source available. Where this really lets things down however is audio generation from the PCM module. That module could really do with the expansion it offers for external DACs over SPI. However, that's a lot of GPIO's and can't offer super-simple, one-wire class-D amplification by strapping a complementary H-bridge driver between an ESP and 8 Ohm loudspeaker!

closed 2019-08-04 by stale[bot]