nodemcu/nodemcu-firmware #2240: Malformed HTTP request produced by http.post() and other http.* methods when URL contains port number

sunlaud, 2018-01-24:
8<------------------------ BUG REPORT -----------------------------------------
### Expected behavior
Correct request should be constructed when sending HTTP requests using methods from [http](https://nodemcu.readthedocs.io/en/master/en/modules/http) module for URLs containing explicit port number (e.g. http://server:port). E.g. POSTing to URL http://example.com:8080/endpoint with empty body should produce the following request:
```
POST /endpoint HTTP/1.1
Host: example.com:8080
Connection: close
```
Checked methods `http.post()`, `http.put()`, `http.request()` and all of them produce malformed request.

### Actual behavior
The request is malformed, coz it contains port number in position, which is treated as HTTP header (with empty name). Here is the example of actual malformed request produced by POSTing to URL http://example.com:8080/endpoint with empty body:
```
POST /endpoint HTTP/1.1
Host: example.com:8080
Connection: close
:8080
```
Note the string `:8080` after "Connection" header. Servers fail to interpret such request.

### Test code
Code to reproduce:
```Lua
http.put("http://example.com:8080/endpoint")
```

#### Workaround code
For now I am using a workaround using raw sockets, which produces correct HTTP request:
```Lua
  sock=net.createConnection()
  sock:on("connection", function(sck)
    sck:send("PUT /endpoint HTTP/1.1\r\nHost: example.com:8080\r\nConnection: close\r\n\r\n")
  end)
  sock:connect(8080, "example.com")
```

### NodeMCU version
NodeMCU custom build by frightanic.com
        branch: master
        commit: 5073c199c01d4d7bbbcd0ae1f761ecc4687f7217
        SSL: true
        modules: adc,bit,cron,crypto,encoder,file,gpio,http,mdns,mqtt,net,node,pwm,rtcmem,rtctime,sjson,sntp,tmr,uart,websocket,wifi,tls
 build  built on: 2018-01-08 20:52
 powered by Lua 5.1.4 on SDK 2.1.0(116b762)

### Hardware
ESP-13 module.

Chip ID       : 12056630
Flash ID      : 1454108
Flash Size    : 4096
Flash Mode    : 2
Flash Speed   : 40000000

8<------------------------ END BUG REPORT -------------------------------------

marcelstoer, 2018-01-25:
Hhmm, with #2202 we see that @paweljasinski was testing against a service on a non-standard port and it all seemed fine (apart from the bug he reported of course). Can you reproduce this with recent build from the `dev` branch?

sunlaud, 2018-01-25:
@marcelstoer thank you for a clue about #2202. In my case the hostname+port combination is exactly 31 chars long so I guess the root cause is the same as in #2202. At least for me it looks like some kind of buffer overflow.
I'll try to test with dev branch.

marcelstoer, 2018-02-04:
Just verified that URLs containing port numbers work just fine.

closed 2018-02-04 by marcelstoer