nodemcu/nodemcu-firmware #2241: BME280 module does not allow coexistence of two BME280s on the same I2C bus

mickley, 2018-01-28:
8<------------------------ BUG REPORT -----------------------------------------
### Expected behavior
The BME280 supports two different I2C addresses, so two BME280s can be used on the same bus.  This should be possible under nodemcu

### Actual behavior
The bme280 module doesn't allow the I2C address to be set and instead looks for devices on the first address, and failing to find one, looks on the second address.  Therefore, if 2 devices are present, only the first can be used.  

It would be better if BME280 functions had an optional address parameter to allow for multiple devices.  

### NodeMCU version
Dev: https://github.com/nodemcu/nodemcu-firmware/commit/ed56d949ee4d00ec10e29ec91d87bcc9b76b9d86

### Hardware
WeMos D1 Mini, with two BME280s connected to the same i2c bus.

8<------------------------ END BUG REPORT -------------------------------------

Fisherman48, 2018-02-16:
Just shift sda - scl wires to the second and call setup
i2c.setup(0, 3, 4, i2c.SLOW)
i2c.setup(0, 4, 3, i2c.SLOW)
I have done this with 6 BME280 using 3 pins (charlieplexing 3x2=6)

closed 2018-02-24 by marcelstoer

mickley, 2018-02-24:
*sigh*.  I'm shocked you're closing this with such a hack solution to a rather simple problem.  The whole point of the I2c bus is to NOT have to do this.  It's an acceptable short-term hack, but not really a fix.

For example, when one has hundreds of setups and only 4 rather long wires leading to the sensors, while it is technically possible to split sensors up and switch the wires, it's really not worth the time.

heikomat, 2020-05-23:
I too am facing the exact same issue, and using the same pin as clock for one and data for the other device seems really hacky to me aswell.

marcelstoer, 2020-05-23:
Yeah, closing it might actually have discouraged some folks from fixing it properly.

TerryE, 2020-05-23:
This isn't a bug per se.  The current behaviour is a functional limitation and removing this is really an enhancement.  If @heikomat or @mickley want to do the work on the module, test and submit as a PR then I will be happy to review and merge it.

heikomat, 2020-05-24:
@TerryE I'll see what i can do 👍

heikomat, 2020-05-24:
I've got it working. Will open a PR soon.

heikomat, 2020-05-24:
Done. @TerryE i'd appreciate a review :)
#3124