nodemcu/nodemcu-firmware #2242: Building luac.cross with LUA_NUMBER_INTEGRAL defined

cwrseck, 2018-01-28:
Make sure you read and understand http://nodemcu.readthedocs.io/en/dev/en/support/.
Use one of the two templates below and delete the rest.

8<------------------------ BUG REPORT -----------------------------------------
### Expected behavior
luac.cross builds correctly.
### Actual behavior
luac.cross fails to build.
### Test code
Provide a [Minimal, Complete, and Verifiable example](http://stackoverflow.com/help/mcve) which will reproduce the problem.
```Lua
lua tools/cross-lua.lua
```
### NodeMCU version
nodemcu-firmware-2.1.0-master_20170824l.
### Hardware
Adafruit HUZZAH
8<------------------------ END BUG REPORT -------------------------------------
Building luac.cross from nodemcu-firmware-2.1.0-master_20170824
fails with the following error (and others later).

cwr@tanner nodemcu $ lua tools/cross-lua.lua
[COMPILE] .build/cross-lua/app__lua__lapi.o
[COMPILE] .build/cross-lua/app__lua__lauxlib.o
[COMPILE] .build/cross-lua/app__lua__lbaselib.o
[COMPILE] .build/cross-lua/app__lua__lcode.o
[COMPILE] .build/cross-lua/app__lua__ldblib.o
[COMPILE] .build/cross-lua/app__lua__ldebug.o
[COMPILE] .build/cross-lua/app__lua__ldo.o
[COMPILE] .build/cross-lua/app__lua__ldump.o
[COMPILE] .build/cross-lua/app__lua__lfunc.o
[COMPILE] .build/cross-lua/app__lua__lgc.o
[COMPILE] .build/cross-lua/app__lua__llex.o
[COMPILE] .build/cross-lua/app__lua__lmathlib.o
app/lua/lmathlib.c:377:43: fatal error: c_limits.h: No such file or directory
compilation terminated.
[builder] Error building target
[builder] Last executed command was: 
gcc -O2 -Iapp/include -Iinclude -Iapp/lua -Wall -DLUA_CROSS_COMPILER -Ddbg_printf=printf -c -MD -MF .build/cross-lua/app__lua__lmathlib.d app/lua/lmathlib.c -o .build/cross-lua/app__lua__lmathlib.o
cwr@tanner nodemcu $ 

The problem has been reported in one form or another for a few years now.
The fix is to ensure that LUA_NUMBER_INTEGRAL is _not_ defined in user_config.h.
A quick fix is to add the specific header paths to cross-lua.lua, using the following
changes to the file:

local cdefs = '-DLUA_CROSS_COMPILER -Ddbg_printf=printf -Dc_strtol=strtol'

local local_include = "-Iapp/include -Iinclude -Iapp/lua -Iapp/libc"

I don't know the full effect of this change.

C W Rose

TerryE, 2018-01-29:
Thanks. I am working on luac.cross at the moment. I will fix this at the same time.

TerryE, 2018-02-13:
I've been doing unit tests on `luac.cross` as part of my LFS work.  In theory,  `luac.cross` should be able to output Integer LC files even if built without `LUA_NUMBER_INTEGRAL` defined, if you use the `-ccn int 32``command-line flags, but this fails in an infinite loop if the Lua code being compiled includes any non-integer numeric constants, where the compiler should just raise a `LUA_ERR_CC_NOTINTEGER` compile error.  

This is all original eLua code so it clearly has never been properly tested. I am not sure whether it is worth sorting out this sort of issue, as the days of Integer builds are numbered.  Lua 5.3 support native integer subtypes and has a lot of other performance goodies, so I am not planning any legacy integer build support.

closed 2018-02-13 by TerryE

TerryE, 2018-03-14:
@cwrseck, See #2292.  This LFS patch brings luac.cross into the core build whether or not LFS is enabled.  It also supports integer builds.