nodemcu/nodemcu-firmware #2245: MQTT BUG??

chathurangawijetunge, 2018-02-02:
hi i am using the code

m = mqtt.Client(node.chipid(), 5, "", "")
m:lwt("/lwt", "offline", 0, 0)
m:on("connect", function(client) print ("connected") end)
m:on("offline", function(client) print ("offline") end)

-- on publish message receive event
m:on("message", function(client, topic, data) 
  print("incomming...",topic .. ":" ) 
  if data ~= nil then
    print(data)
  end
end)

function connect()
  print("connecting.....")
  m:connect("broker.hivemq.com", 1883, 0, function(client)
    print("connected")

    client:subscribe("/sw1", 0, function(client) print("subscribe success") end)
    client:publish("/sw1", "hello", 0, 0, function(client) print("sent") end)
  end,
  function(client, reason)
    print("failed reason: " .. reason)
    tmr.alarm(6,2000, 0, function() connect() end)
  end)
end

connect()
-----------------------------------------------------

this works fine... 
until i remove the phone line from my rougher and re connects after some time this code re connects to the MQTT broker.
but after 5 seconds it fires the offline function and reconnects, this happens every 5 seconds there after...
is it a bug????

FrankX0, 2018-02-02:
Sounds like a duplicate of #2197.

chathurangawijetunge, 2018-02-02:
So is it a bug ?

FrankX0, 2018-02-02:
In my opinion: yes.

chathurangawijetunge, 2018-02-04:
So how can we sort out this issue (bug)

closed 2018-02-04 by marcelstoer

chathurangawijetunge, 2018-02-04:
Will this be fix any time sooner