nodemcu/nodemcu-firmware #2247: enduser setup feature

davidcbittner, 2018-02-02:
Additional parameter during AP setup - Wi-Fi sethostname

Justification
being able to set a host name during the initial AP connection and setup is key to being able to identify the module on the STA Wi-Fi network after it successfully connects.  This is especially useful if the unit is assigned an IP address from the DCHP. In addition storing this information during the initial AP setup and subsequent AP resets eliminates the need for a secondary setup with additional coding and embedded web pages for additional setup step to allow the end user to set this information. By eliminating the secondary setup we free up storage/memory on the device. Having unique host names such as "bedroom_1", "kitchen" or "weather_station" is more manageable for the non-techy person who may need to do some additional setup on their home router/access point to allow DNS resolution or port configuration on their network.  

Workarounds
1. Manually set host name after initial AP setup has occurred with hard coded prefix concatenated with a unique identifier such as node.chipid() or something else to make it unique.  
2. Additional web page served to set up and store additional parameters such as the host name, obtain IP address from DHCP or assign IP address manually through additional setup configuration web page.

[image]

This is my first time posting a additional feature on github I believe this is the correct format. I am not sure about the attached image of the additional feature I mocked up I hope this attaches correctly.
Regard
David

marcelstoer, 2018-02-04:
There is a pending PR #2132 which will allow for arbitrary extensions to the available parameters. Once it's merged it you should be able to implement what you need here quite easily, no?

davidcbittner, 2018-02-04:
Yes I believe so. 

Thanks

David

closed 2018-02-04 by marcelstoer