nodemcu/nodemcu-firmware #2253: DHT version detection, DHT12 and "misbehaving" DHT11

orcuspl, 2018-02-04:
I have just received a small batch of DHT sensors from China. They look like and are marked as DHT11, but looking at different source codes they behave like DHT12 - they send fractional temperature. 

In current implementation using dht.read11() fails due to checksum calculation. Using dht.read() and dht.readxx() gives incorret results as data format is different.

Fixing checksum calculation is quire easy, but there is also something fishy in current implementation of dht.read(). There are 2 data formats it uses to interpret data:
1. DHT11 format:
```
dht_humidity    = dht_bytes[0];  // dht_bytes[1] == 0;
dht_temperature = dht_bytes[2];  // dht_bytes[3] == 0;
```
2. "Non-DHT11" format:
```
#define COMBINE_HIGH_AND_LOW_BYTE(byte_high, byte_low)  (((byte_high) << 8) | (byte_low))
dht_humidity = (double)COMBINE_HIGH_AND_LOW_BYTE(dht_bytes[0], dht_bytes[1]) * 0.1;
dht_temperature = (double)COMBINE_HIGH_AND_LOW_BYTE(dht_bytes[2] & 0x7F, dht_bytes[3]) * 0.1;
```
It looks good until we look how dht.read()/dht_read_universal() determines which format should be used. It is done by checking: `(dht_bytes[1] == 0) && (dht_bytes[3] == 0)`. If this is true then is assumes DHT11 and uses format 1. Otherwise it uses format 2. This will return wrong results for some cases:
1. When something other than DHT1x actually senses value that have zeros in "fractional" parts. For example 25.6C will result in dht_bytes[2] = 1 and dht_bytes[3] = 0. There are only few combinations of temperature and humidity that will trigger this, but I think we shouldn't ignore them.
2. In case of DHT12 and my "misbehaving" DHT11 format 2 will be used.

I'm considering learning how to send pull requests myself (if nobody else picks it up), but I have no experience and limited access to hardware (only those DHT11). I would like to get any feedback on some solutions:
1. Fix checksum calculation and incorporate fractional part of results for DHT1x. I think about introducing read1x function and mapping read11 and read12 to it. It shouldn't break normal DHT11 as in the results fractional parts are always 0 for DHT11.
2. Rare detection issues. I was thinking about two possible solutions but neither feels good:
a. Ignore it and document as known issue if anybody will be so unlucky to hit it. 
b. Remove universal function and force users to decide which DHT version and format they will use. It will break backward-compatibility. Don't know if probability of the issue is high enough to justify it. 

I'm especially interested in some option 2.c. that could force changing approach for the checksum issue.

lou-lan, 2018-02-20:
Hello, I also encountered this problem, hope to repair.

lou-lan, 2018-02-23:
I found new DHT11, DHT11, DHT12, They are different.

starDemo, 2018-05-08:
The new version dht11 has different data format, waitting to upgrade

closed 2019-08-04 by stale[bot]