nodemcu/nodemcu-firmware #2257: mqtt:connect callback issue

DavidBurg, 2018-02-11:
### Expected behavior
Only one of the two callback functions for mqtt:connect should only be called

### Actual behavior
This is occurring when using a secure connection.  The following sequence occurs:

- call mqtt:connect
- the connection is established and the appropriate callback is executed (function mqtt_connected())
- later, stop the MQTT broker (systemctl stop mosquitto.service)
- the "connection could not be established" callback (function mqtt_error()) is unexpectedly executed, even though mqtt:connect was not called again

notes:

- The deprecated "autoreconnect" argument to mqtt:connect is omitted from the call, and defaults to 0.
- mqtt:connect is only called once.  Its "success" callback is called twice.  Later its "fail" callback is called.
- A similar behavior was noted by @andreykeen in #2197.  Andrey referenced my [StackOverflow question](https://stackoverflow.com/questions/48335112/mqtt-clientconnect-not-established-callback-is-unexpectedly-called-after-a?noredirect=1#comment83656762_48335112), where I provided trace information and a theory of why this is happening
- My test code (below) is adapted from Andrey's code in #2197.  His code has an mqtt.client:on callback for "offline", which should be needed to reestablish the MQTT connection (as in the example in the documentation).  However, due to this bug, the MQTT connection will be reestablished without this callback.  So, I removed it from the example code.  I also changed Andrey's code to create a secure connection in the "m:connect" call.

### Test code
```Lua
local timer_mqtt_connect = tmr.create()

local m = mqtt.Client(MQTT_CLIENT_ID, 20, MQTT_USER, MQTT_PASS, 1)

-- removed m:on("offline", function(client)
-- removed     print ("MQTT: Broker going to offline")
-- removed     client:close()  -- I tried with and without it
-- removed     timer_mqtt_connect:start()
-- removed end)

local function mqtt_connected(client)
    print("MQTT: Connected")
end

local function mqtt_error(client, reason)
    print("MQTT: Connecting failed. Reason:"..reason)
    -- removed client:close()  -- I tried with and without it
    timer_mqtt_connect:start()
end

local function do_mqtt_connect()
    print("MQTT: Trying to connect. Heap:"..node.heap().." Uptime:"..tmr.time())
    m:connect(MQTT_HOST, 8883, 1, mqtt_connected, mqtt_error)
end

timer_mqtt_connect:register(5*1000, tmr.ALARM_SEMI, function()
    do_mqtt_connect()
end)

timer_mqtt_connect:start()
```
### Trace
```
enter mqtt_socket_client.
NodeMCU_66f13e
length username: 12
length password: 12
MQTT: Init info: xxx, xxx, xxx
leave mqtt_socket_client.

-- mqtt:connect is called here
MQTT: Trying to connect. Heap:31008 Uptime:97
enter mqtt_socket_connect.
TCP ip is set: 10.0.0.100
TCP port is set: 8883.
enter socket_connect.
leave socket_connect, heap = 28160.
leave mqtt_socket_connect.
enter mqtt_socket_connected.
Send MQTT connection infomation, data len: 54, d[0]=16 
leave mqtt_socket_connected.
enter mqtt_socket_sent.
enter mqtt_socket_received.
MQTT: Connected -- mqtt:connect's "success" callback is executed twice
MQTT: Connected

send_if_poss, queue size: 0
leave mqtt_socket_received.
enter mqtt_socket_timer.
timer, queue size: 0
keep_alive_tick: 1
leave mqtt_socket_timer.
enter mqtt_socket_timer.
timer, queue size: 0
keep_alive_tick: 2
leave mqtt_socket_timer.
enter mqtt_socket_timer.
timer, queue size: 0
keep_alive_tick: 3
leave mqtt_socket_timer.
enter mqtt_socket_timer.
timer, queue size: 0
keep_alive_tick: 4
leave mqtt_socket_timer.
enter mqtt_socket_timer.
timer, queue size: 0
keep_alive_tick: 5
leave mqtt_socket_timer.
enter mqtt_socket_timer.
timer, queue size: 0
keep_alive_tick: 6
leave mqtt_socket_timer.

-- Shut down the mqtt broker with "systemctl stop mosquitto.service"
enter mqtt_socket_reconnected.
MQTT: Connecting failed. Reason:-5  -- mqtt_error() is unexpectedly executed
enter mqtt_socket_disconnected.
leave mqtt_socket_disconnected.
leave mqtt_socket_reconnected.

-- Since mqtt_error() was called above, timer_mqtt_connect is
-- restarted, causing do_mqtt_connect() to be called.

MQTT: Trying to connect. Heap:30848 Uptime:108
enter mqtt_socket_connect.
TCP ip is set: 10.0.0.100
TCP port is set: 8883.
enter socket_connect.
leave socket_connect, heap = 28296.
leave mqtt_socket_connect.

enter mqtt_socket_reconnected.
MQTT: Connecting failed. Reason:-5
enter mqtt_socket_disconnected.
leave mqtt_socket_disconnected.
leave mqtt_socket_reconnected.
```

### NodeMCU version
dev branch, pulled 2018-02-10.  It includes the @FrankX0 1-line fix to mqtt.c for #2256

### Hardware
NodeMCU dev kit

FrankX0, 2018-02-11:
> mqtt:connect is only called once. Its "success" callback is called twice.

Actually this is not true. The debug output of the C-code (`MQTT: Connected`) happens to be the same as used in the Lua code. Please check.
I will look into the other (unexpected) behaviour.

DavidBurg, 2018-02-12:
Agreed.  I didn't notice that the two messages were the same.  Thanks!

FrankX0, 2018-02-12:
@DavidBurg: please give the referenced pull request a try.
After a successful connection, only the "offline" function is called.
I have tested this with an standard connection. If possible please give it a try with a secure connection.

DavidBurg, 2018-02-13:
@FrankX0: I tested your fix with a secure connection.  It worked as expected (i.e., as documented).  Thank you!

PR #2262 (2018-02-13): Disable MQTT connect error callback after successful connection
Fixes #2257 

- [x] This PR is for the `dev` branch rather than for `master`.
- [x] This PR is compliant with the [other contributing guidelines](https://github.com/nodemcu/nodemcu-firmware/blob/dev/CONTRIBUTING.md) as well (if not, please describe why).
- [x] I have thoroughly tested my contribution.
- [x] The code changes are reflected in the documentation at `docs/en/*`.

Disable the error callback of mqtt.client:connect() after a successful connection.
This will prevent this function to be called after a future disconnection.
 [... cut]
closed 2018-02-13 by marcelstoer