nodemcu/nodemcu-firmware #2258: Cant' autodetect firmware

Schwenkner, 2018-02-11:
Can someone give me a hint what's going wrong?
I've several Wemos D1 pro and it works well with the nodemcu firmware.
If I boot the device, I get

> Can't autodetect firmware, because proper answer not received (may be unknown firmware). 
> Please, reset module or continue.
> load 0x40100000, len 27228, room 16 
> tail 12
> chksum 0x92
> ho 0 tail 12 room 4
> load 0x3ffe8000, len 2184, room 12 
> tail 12
> chksum 0xfd
> ho 0 tail 12 room 4
> load 0x3ffe8888, len 136, room 12 
> tail 12
> chksum 0x01
> csum 0x01
> rf cal sector: 4091
> freq trace enable 0
> rf[112] : 00
> rf[113] : 00
> rf[114] : 01
> 

which seems to be ok since the device works.

But one of mine devices which did work already now boots with

> Can't autodetect firmware, because proper answer not received (may be unknown firmware). 
> Please, reset module or continue.
> load 0x40100000, len 26820, room 16 
> tail 4
> chksum 0x39
> load 0x3ffe8000, len 2256, room 4 
> tail 12
> chksum 0x4e
> ho 0 tail 12 room 4
> load 0x3ffe88d0, len 136, room 12 
> tail 12
> chksum 0x51
> csum 0x51
> need boot 1.4+
 
I've flashed with the same firmware.

Do I have a chance to reactivate this device?
What does "need boot 1.4+" mean?
Can I reset the device anyhow?

devsaurus, 2018-02-11:
> I've flashed with the same firmware.

**How** did you flash **which version** of firmware?

> Do I have a chance to reactivate this device?

Yes, by erasing the flash chip completely and re-flashing the firmware.
For instructions see [Flashing the firmware](http://nodemcu.readthedocs.io/en/dev/en/flash/), especially upgrading](http://nodemcu.readthedocs.io/en/dev/en/flash/#upgrading-firmware).

Schwenkner, 2018-02-11:
Thx for the answer. I'm using a nodemcu version build with the cloud build service. That's I think it is the newest version. What I didn't write, I have had running the same nodemcu version already on that device. Then I was running my lua program on battery supply using deep sleep. I would check how long it works. The program did run for about 4 weeks then the battery was going low and it stopped running. After powering the device then after via USB the problem happened. I cannot really imagine that this is the reason. And after trying to reflash the same nodemcu build, I'm getting the above output. Flashing it on another device it works, thus I cannot imagine having a wrong nodemcu build.

I've also tried =file.format() without success.

According to your hint I've now tried to using SDK init data description. Flashing the file "esp_init_data_default.bin" to address 0xffc000 (I have a Wemos D1 pro) does not work, it ends with an error without any hint. I'm not using the Python tool, but the ESP8266Flasher.exe, since I haven't installed Python yet.

Is the Python tool worth to try?

devsaurus, 2018-02-11:
Don't bother with esp_init_data_default.bin, it's supplied by the firmware image.
If you haven't installed python I recommend [NodeMCU PyFlasher](https://nodemcu.readthedocs.io/en/dev/en/flash/#nodemcu-pyflasher). Just make sure that "Erase flash" is enabled.

Schwenkner, 2018-02-11:
Thx, but....
I've tried. With "Erase flash" it hangs here:

> Connecting....
> Detecting chip type... ESP8266
> Chip is ESP8266EX
> Uploading stub...
> Running stub...

How long do I have to wait, did it 10 minutes.

When I then flash nodemcu, this is the output

> Connecting....
> Detecting chip type... ESP8266
> Chip is ESP8266EX
> Uploading stub...
> Running stub...
> Stub running...
> Configuring flash size...
> Auto-detected Flash size: 16MB
> Flash params set to 0x0090
> Compressed 456352 bytes to 294574...
> Wrote 456352 bytes (294574 compressed) at 0x00000000 in 40.1 seconds (effective 91.0 kbit/s)...
> Hash of data verified.
> 
> Leaving...
> 
> Done.

But nothing changed, still getting:

> Can't autodetect firmware, because proper answer not received (may be unknown firmware). 
> Please, reset module or continue.
> load 0x40100000, len 27228, room 16 
> tail 12
> chksum 0x92
> ho 0 tail 12 room 4
> load 0x3ffe8000, len 2184, room 12 
> tail 12
> chksum 0xfd
> ho 0 tail 12 room 4
> load 0x3ffe8888, len 136, room 12 
> tail 12
> chksum 0x01
> csum 0x01
> need boot 1.4+

Some more ideas? Won't give up.

Schwenkner, 2018-02-11:
Sorry for closing, was a mistake.

Schwenkner, 2018-02-11:
Ok, next try was with "Erase flash" AND the nodemcu file name given.
This did the job, chip works now.

Thx

closed 2018-02-11 by Schwenkner

filozof71, 2018-03-14:
Hello,

Did you mean somethinga like this:

python esptool.py --port COM2  erase_flash d:\nodemcu\cloud_firmware_build\nodemcu-master-16-modules-2018-03-12-03-22-25-integer.bin

I am nodemcu Amica (https://www.gearbest.com/transmitters-receivers-module/pp_366523.html). I booght two and none works. I came here due to "load 0x40100000, len 27228, room 16" search in Google. whever I flashed my board it always gives some garbage signs. Lately I tried on 74880 speed and I obtained this error. I followed this thread and performed "earse_flash" but it have not helped. The command with file name is wrong.

This is my first ESP8266. I working with Raspberry Pi, and earlier on big servers systems in C++ and Java. 

Any help will be appreciated.

best regard Olek

D:\nodemcu\nodemcu-pyflasher>python esptool.py --port COM2  --baud 74880 erase_flash
esptool.py v2.2.1
Connecting....
Detecting chip type... ESP8266
Chip is ESP8266EX
Uploading stub...
Running stub...
Stub running...
Erasing flash (this may take a while)...
Chip erase completed successfully in 2.0s
Hard resetting...

D:\nodemcu\nodemcu-pyflasher>python esptool.py --port COM2  --baud 74880 write_flash -fm qio 0x00000 d:\nodemcu\cloud_firmware_build\nodemcu-master-16-modules-2018-03-12-03-22-25-integer.bin
esptool.py v2.2.1
Connecting....
Detecting chip type... ESP8266
Chip is ESP8266EX
Uploading stub...
Running stub...
Stub running...
Configuring flash size...
Auto-detected Flash size: 4MB
Flash params set to 0x0040
Compressed 651856 bytes to 429124...
Wrote 651856 bytes (429124 compressed) at 0x00000000 in 57.1 seconds (effective 91.4 kbit/s)...
Hash of data verified.

Leaving...
Hard resetting...