nodemcu/nodemcu-firmware #2263: enduser_setup new feature

davidcbittner, 2018-02-13:
### Missing feature
Displaying IP address assigned after succesful AP configuration and IP address assigned.

### Justification
For ease of setup and use without having to rummage around to find out what the IP address of the unit is after you have gone through a successful AP configuration and connected to a WIFI network. Currently without having the hostname additionally as a parameter to set during the AP configuration a user has to search their wifi router to find out what has been assigned. This is time consuming and a nuisance. Having the IP address appear on the "Success!" screen once it connects would be a trememdous benefit. It's impossible to write instructions to an end user on how they can obtain their IP address on so many brands, makes and models of wifi routers and AP point forget it you can't do it and give screen shots.

### Workarounds
I am not sure there is one.
[image]

FrankX0, 2018-02-13:
That's what I had implemented almost a year ago: #1867

davidcbittner, 2018-02-13:
This is probably a dumb question however I just flashed my unit with the latest 1.5.4.1 should I not see it in there?

davidcbittner, 2018-02-13:
sorry I closed this by mistake

FrankX0, 2018-02-13:
Firmware version 1.5.4.1 is old.
Please generate either a current master or dev build, using https://nodemcu-build.com

davidcbittner, 2018-02-13:
perfect thanks very much.

closed 2018-02-13 by davidcbittner