nodemcu/nodemcu-firmware #2264: rtctime problem

Schwenkner, 2018-02-13:
I'm playing with rtctime.dsleep and time settings. To use rtctime.dsleep I have to set rtctime before calling sleep. Therfore I use sntp.sync. This works fine.

But how I understood the documentations, the time should be kept after awake from deep sleep. This doesn't work for me, time is always 1/1/1970. according to documentation it is necessary to re-sync at least once after deep sleep. I'm doing this several time without success.

Yes, I could call sync every time after awake, but I will save some power when it is not necessary to call sync. It would be fine for me to call it only once a day.

Here is my code I'm trying:
`
function sleep()
	print("--sleep")
	tmr.create():alarm(1000, tmr.ALARM_SINGLE,
		function()
			rtctime.dsleep_aligned(10*1000000, 10*800000)
		end
	)
end

function application.run()
	print("-- running")
	local tm = rtctime.epoch2cal(rtctime.get())
	print(string.format("-- %04d/%02d/%02d %02d:%02d:%02d", tm.year, tm.mon, tm.day, tm.hour, tm.min, tm.sec))
	if tm.year == 1970 then
		print("-- have to sync")
		sntp.sync(
			function(time)
				sleep()
			end
		)
	end
end
`
Is there any working sample how to fetch the time after deep sleep? I couldn't find anything.

pjsg, 2018-02-14:
What version of nodemcu are you using? This sounds very similar to the issue that was fixed in https://github.com/nodemcu/nodemcu-firmware/pull/2148 in last october. 

Actually it appears that we haven't done a master drop since then..... Hmm....

To get a fixed version, you want to build off 'dev'

Schwenkner, 2018-02-14:
I'm using master branch, will give the dev branch a try, thx

marcelstoer, 2018-02-14:
> Actually it appears that we haven't done a master drop since then

Nope, we were waiting for the KRACK fix from Espressif.

closed 2018-02-24 by marcelstoer