nodemcu/nodemcu-firmware #2265: Clarify gpio.pulse.build documentation

marcelstoer, 2018-02-15:
https://stackoverflow.com/q/48790455/131929 suggest that `gpio.pulse.build` could do with some additional explanation. Can't judge this myself, though.

pjsg, 2018-02-16:
@marcelstoer It turns out that the bug (I think) is that 

```
gpio.mode(1, gpio.OUTPUT)
```

does *not* actually make the output pin live. That (currently) happens on the write gpio.write to that pin. As far as I can tell, this is really a bug, and setting the mode ought to turn on the pin driver (with or without open drain as appropriate). I can certainly fix this in my current gpio.pulse pull request. Alternatively, I can enable the output drivers at the start of the pulse output -- but that seems rather odd.

I'll also add some more examples.

HHHartmann, 2018-02-16:
To me it was not clear that loop referenced another step. So detailing that it is a jump to another step and maybe a dedicated sentence to numbering of the steps (starts by 1) would be good. But then again:that's lua Standard. 
Also describing what the example does world IMHO help more than adding more examples. (Gave me the clue to understand the mechanism)

closed 2018-02-23 by marcelstoer