nodemcu/nodemcu-firmware #2266: socket event handler not cleaned when closing from another socket's handler

netadair, 2018-02-15:
Hello,
I'm currently writing a telnet/uart server (also somewhat RFC2217-compliant). Everything works fine after solving a number of memory constraints, two tcp servers sitting and handling requests, with the send data queued etc.

As I can have only one uart connection I need to close a possible lurking old connection to have the new one talk to the uart. Similar, I only want one telnet session on the box, to not multiplex output.

When a session is closed from the client side, everything works fine and as expected.

However if I close the one tcp socket from a another (freshly created) one, addressed via a global variable holding the "old" socket, about 512 bytes of memory leaks.
After finding the helpful debug.getregistry() function, I learned three functions stay in the memory, turned out these are event handler of the old socket.

As a workaround,  I reset the old socket's event handler from the new socket:

```Lua
 [...]
     uart_srv:listen(UART_PORT, function(c)

             local fifo = {}
             local fifo_drained = true
     
             if global_c~=nil then   -- schliesse alte verbindung
 
                 global_c:on("receive", nil)
                 global_c:on("sent", nil)

                 local c_c=global_c
                 global_c:close()
 
                 c_c:on("disconnection", nil)
                 c_c=nil
 
             end
             global_c=c
 [...]
```

Any idea why this happens?

netadair, 2018-02-15:
PS: So there was a little test case missing, at first I did my tests only with "empty" sessions, just opening but sending (and receiving) nothing.
As soon as data is sent over the socket, the three event handler just stay referenced and leak memory after the socket is closed.

pjsg, 2018-02-18:
I would suspect that references to the old socket are being held somewhere, and this holds onto the event handlers.

nwf, 2018-03-23:
If memory serves, when I was writing the hack that is https://github.com/AcmeTensorToys/esp-lua_core/blob/master/telnetd/telnetd.lua , I, too, needed to explicitly `nil` out the callbacks as well as `:close()` the socket.  I think this is because, in my case, the callbacks were holding on to objects that, in turn, held on to the socket, creating a nice circular subgraph.  I did not investigate beyond that, tho', and would be curious to know if there's a more subtle bug (e.g. someone holding a strong reference when they meant to be holding a weak one?).

TerryE, 2018-04-11:
> Any idea why this happens?
Do the checks: e.g. 
```Lua
  for k,v in pairs(_G) do print (k,v) end
  for k,v in pairs(debug.getregistry()) do print (k,v) end
```
the old socket might be listed directly as a user data, or hidden as an upval in a closure (function).

Whatever, this issue isn't a valid on for raising here.  See our support guidelines.

closed 2018-04-11 by TerryE