nodemcu/nodemcu-firmware #2268: Some Question In Sleep Function

grey-yibu, 2018-02-20:
I'm a fans of NodeMcu .Recently I'm doing some optimization to lower the power .I have check the NodeMcu API and it seems like there is 'node.sleep()' ,'node.dsleep()' ,'wifi.suspend()' and 'wifi.resume()' can be use. But when I test my NodeMcu ,it seems like only node.dsleep() can work. When I use the other function, the NodeMcu studio says that  'stdin:1: attempt to call field 'sleep' (a nil value)'.

Can any body give me some suggest?

FrankX0, 2018-02-20:
`node.sleep()` only gets compiled if `PMSLEEP_ENABLE` is defined in user_config.h
By default this is disabled.

marcelstoer, 2018-02-20:
Thanks Frank.

https://nodemcu.readthedocs.io/en/latest/en/modules/node/#nodesleep

closed 2018-02-20 by marcelstoer

grey-yibu, 2018-02-23:
Thanks a lot……