nodemcu/nodemcu-firmware #2270: esp32 make flash error:patition data subtypes 'rf' is not valid

kobelan, 2018-02-22:
esp32 make flash error:patition data subtypes 'rf' is not valid
[image]
[image]
[image]

devsaurus, 2018-04-05:
@jmattsson there's still the custom partition table around at `components/platform/partitions-2MB.csv`. As far as I can see it's outdated as stated above and not used anywhere. I can't remember what the initial purpose of this was. So far I understood that spiffs searches for partition type 0xC2 and appends a new partition for spiffs if it can't find one.
Should we update it to match current IDF format or should we get rid of it?

jmattsson, 2018-04-05:
Oh gosh, I haven't been in ESP32-land for so long now!

Umm... I think we had the custom table to get the rfdata into a known spot? I'm very much not across the changes to the IDF, so if that's no longer an issue I think we could probably get rid of the custom partition table. Just remember to remove any reference to it from the idf-default overrides.

devsaurus, 2018-04-15:
I've fixed and documented the template partition table with https://github.com/nodemcu/nodemcu-firmware/commit/45be28cf60901513ed8ec72bb66bd8db36240d7a.

closed 2018-04-15 by devsaurus