nodemcu/nodemcu-firmware #2274: Module ds18b20 determining the number of detected sensors

tjakober, 2018-02-25:
8<------------------------ FEATURE REQUEST ------------------------------------
### Missing feature
In module ds18b20 there is no information returned about how many sensors are detected which will call the read callback function.
### Justification
Using the suggested programming model one cannot determine how many devices of temperature sensors are readout during a readout call. This is of course not required in the example where its only printed on the serial port. However when you want for example to send the complete set of results at once via internet one must know when all the devices are read out to take the further actions.

It would be easiest to add a flag to the callback function to indicate whether it is the last device which was readout.

Another possibility would be in returning the number of devices detected when calling the setting function. However this would need more logic within the callback function to start the further actions and will not be safe if one device did not answer during the read.

### Workarounds
I was using a "software soldering iron": Setup a timer which ensures all devices are readout and then take the further actions.
8<------------------------ END FEATURE REQUEST --------------------------------

Thanks for taking care of this issue
Thomas Jakober

TerryE, 2018-03-02:
@tjakober Thomas,  We don't have an active maintainer for this module.  The original contributer was @fetchbot and the main reviewer was @nickandrew.  See #2003 for the backstory.  No point in repeating here.  

I  use my own version of the Lua module because it reads faster and is functionally richer including the stuff that you are looking for, but what get committed here is by the majority consensus.

However, if you want to make the changes yourself and raise a PR then we'll take a look.

HHHartmann, 2019-07-21:
This module had been replaced by a lua version

closed 2019-07-21 by HHHartmann